iNavFlight/inav #5316: Is any utility like Pidtoolbox available for iNav?

Improg, 2020-01-06:
I need to view response charts for my quad. Pid Analyzer don't work with iNav 2.3.0, flying iNav 2.1.0 some strange errors occurs, so I can't set up pids for old version instrumentally and then change to new one. Pidtoolbox don't work with 2.3.0 too as with 2.1.0 don't work. How can I get response charts (roll, pitch, yaw) for the latest iNav 2.3.0?
If no exist, is there an ability to convert blackbox from 2.3.0 format down to 2.1.0 to use Pid Analyzer? Or, convert to betaflight format for Pidtoolbox?

DzikuVx, 2020-01-06:
https://github.com/Plasmatree/PID-Analyzer works

closed 2020-01-06 by DzikuVx

Improg, 2020-01-06:
Don't work with 2.3.0 log.

DzikuVx, 2020-01-06:
Tested last week and worked nicely

DzikuVx, 2020-01-09:
Try this version: https://github.com/iNavFlight/blackbox-tools/releases

Improg, 2020-01-10:
Checked - I used 0.4.3 version. Now it works with 0.4.4. Thanks!