iNavFlight/inav #5321: Major Issue - Matek F411 Wing with Matek F411 RSSI Board Target

Zangetsu57, 2020-01-11:
Hello,

First off, this is the first time I've used the Matek F411 wing, but I've used the Matek F405 Wing in several builds.  With iNav 2.3, it appears that when the Matek F411 RSSI target is selected, the board often doesn't boot properly when powered on.  It seems to boot only around 1 out of 10 times.  This is evident as a camera video feed is visible, but the OSD never appears.  Additionally, servo input does not work.

I tried the Matek F411 target with iNav 2.3, and it boots on a more consistent basis, but failures do occur.  

I scaled back to iNav 2.2.1 with the Matek F411 RSSI target.  It works more consistently than iNav 2.3, but boot failures still occur.

It's possible that I have a defective board, but don't have another one to try.

giacomo892, 2020-01-11:
What are the LEDs on the board showing when it is not booting? Do they have a pattern?
https://github.com/iNavFlight/inav/wiki/Blinkenlights

Zangetsu57, 2020-01-11:
Thanks for the reply.  They do blink, but I don't recall the pattern.  I'll check tomorrow and report back.

Jetrell, 2020-01-11:
@Zangetsu57 What you have described sounds like an issue that is still present with the F411-wing firmware.    
Make sure - `osd_video_system = AUTO` 
Do not have it set to- ` PAL` or `NTSC`

Zangetsu57, 2020-01-11:
Thanks Jetrell, I made the change to AUTO and can confirm that the problem described above is resolved for both iNav 2.2.1 and 2.3!

closed 2022-02-10 by DzikuVx