iNavFlight/inav #5324: OMNIBUSF4V3 (Banggood clone) external BARO not working

ratmole, 2020-01-13:
on the latest inav (2.3.0) OMNIBUSF4V3 target,the I2C external BARO is not working.
There is a SPI /I2C conflict, on the target.h
see here:
https://github.com/iNavFlight/inav/blob/debe8ab772dc7e13d6cecd162ae107760adfd90c/src/main/target/OMNIBUSF4/target.h#L116
and here:
https://github.com/iNavFlight/inav/blob/debe8ab772dc7e13d6cecd162ae107760adfd90c/src/main/target/OMNIBUSF4/target.h#L119

In my case i wanted a I2C BARO, so had to recompile inav with #define BMP280_SPI_BUS commented out: 
`// #define BMP280_SPI_BUS`

This is the FC used:
https://www.banggood.com/9_4g-30_5x30_5mm-Omnibus-F4-Flight-Controller-AIO-OSD-5V-BEC-and-Current-Sensor-p-1226610.html?rmmds=buy&cur_warehouse=CN

hali9, 2020-01-15:
https://github.com/iNavFlight/inav/pull/5187 should fix this.

hali9, 2020-01-16:
I found that if baro on spi is defined then baro on i2c is ignored:
common_hardware.c 
```
#if defined(BMP280_SPI_BUS)
  BUSDEV_REGISTER_SPI(busdev_bmp280, DEVHW_BMP280, BMP280_SPI_BUS, BMP280_CS_PIN, NONE, DEVFLAGS_NONE);
  #elif defined(BMP280_I2C_BUS) || defined(BARO_I2C_BUS)
    ...
    BUSDEV_REGISTER_I2C(busdev_bmp280, DEVHW_BMP280, BMP280_I2C_BUS, 0x76, NONE, DEVFLAGS_NONE);
#endif
```

closed 2020-11-30 by DzikuVx