iNavFlight/inav #5327: Invert Motor Output

monoxide13, 2020-01-15:
## Current Behavior
Currently it is impossible to invert the output of the motor like the servos can be. Current work around is to use stabilized throttle servo mix and a servo output.

## Desired Behavior
Toggle switch inverts motor mix after flight controller processing (cruise modes, climb/decent power adjustments) and prior to output stage.

## Suggested Solution
Toggle switch added on mixing page to invert throttle output, operating the same way as the servo invert toggle.

## Who does this impact? Who is this for?
All airplane users who use motor mix for throttle and run gas/nitro engines.

## Additional context
I am using the MATEK722SE target, and use all 6 servo outputs and 1 motor output for a nitro engine. The motor servo runs backwards, so stick all the way down actually opens the carb. The short term solution I am going to recompile the target with all 7 servo outputs.

DzikuVx, 2020-01-15:
Motor outputs are designed for electric motors, not driving throttle on nitro/gas engines. 
Nitro/gas throttle servo should be configured as a servo.

Woga65, 2020-01-16:
@monoxide13 
A while ago I needed the same feature to setup various variable pitch vehicles so I added some lines of code to achieve the desired behavior. https://github.com/iNavFlight/inav/pull/4889

I did not take the time to fully implement a new vehicle type and rather implemented it as a quick hack so it did not make it into any release.

Meanwhile the feature is fully tested and works well. You can find it here:
https://github.com/Woga65/inav/tree/2.2.x-vari_pitch

Despite the name „2.2.x-vari_pitch“ it is based on the current INAV >=2.3 master branch.

Binary to test if it meet your needs:
https://github.com/Woga65/inav/releases

monoxide13, 2020-01-18:
Ok. I'll try and find a place to add to the documentation to make a note for others like me. I also see you made it work here https://github.com/iNavFlight/inav/issues/4840 @DzikuVx. Thank's for your efforts.

closed 2020-01-18 by monoxide13