iNavFlight/inav #5330: RTH message is missing after completing waypoint mission

haybeseret, 2020-01-15:
## Current Behavior
On a waypoint mission, when RTH is selected at the end of a mission, the OSD is not showing RTH once the mission ends
Adding to the above, with no clear indication on the mission progress you cannot really tell when your mission is over.
Adding to that when mission is over and craft is in loitr there is no indication on that state as well (i,.e that the mission is over)

## Steps to Reproduce

1. Configure mission
2. Select the Checkbox "RTH at the end of mission"

## Expected behavior
Change the osd from "en route to waypoint" to RTH when the mission is over (and the RTH option is selected as stated above)

## Suggested solution(s)
Change the osd from "en route to waypoint" to RTH when the mission is over (and the RTH option is selected as stated above)

version is Inav 2.2.1

- FC Board name and vendor: Omnibus f4 AIO

hali9, 2020-01-17:
This is because when WPRTH not landing go to state WAYPOINT_FINISHED.
When WPRTH have set land than rth landing is called and when landing success then go also to WAYPOINT_FINISHED. So it's not clear what to enter "Landing" or "Loitering" and it's "en route to wp".

So better change WAYPOINT_FINISHED mwState
from `MW_NAV_STATE_WP_ENROUTE` "EN ROUTE TO WAYPOINT"_
to `MW_NAV_STATE_HOVER_ABOVE_HOME` "HOVERING/LOITERING AROUND HOME"
and
not call navOnEnteringState_NAV_STATE_RTH_FINISHING in navOnEnteringState_NAV_STATE_WAYPOINT_RTH_LAND
but switch to it.

peteoz, 2020-03-02:
similarly during WP Mission there is no indication what WP number the model is approaching or has already passed, can this be modified to display "EN ROUTE TO WAYPOINT 21" (or "EN ROUTE TO WP21") when the model is moving towards WP no. 21 (as an example) ?

breadoven, 2021-11-29:
Closed by:

https://github.com/iNavFlight/inav/pull/6979
https://github.com/iNavFlight/inav/pull/6524

closed 2021-11-29 by breadoven