iNavFlight/inav #5332: Problems with altitude hold

tibean, 2020-01-17:
Hi.

First of all I would like to give you my congratulations for the wonderful software that you developed.
But after several months of frustration with Altitude Hold feature, I’m asking for help. Not sure if this is the current place, let me know if there is a better one.
I have a 5 inch quad with a Matek F405STD, equipped with a sonar TFMini pointing to the ground. I’ve used the Present setting for this type of quad.
There are two major issues that I’m not able to overcome:
1.	Quad is not able to keep altitude, with stock PID values, or even after a (tentative) PID tuning to the Altitude Hold parameters. When hovering two meters from the ground, the quad slowly starts to drop altitude and hits the ground. And when hovering, if I softly move the quad forward (pitch), it will lose altitude and hit the ground.
2.	Sonar seems not be doing nothing, ie, when I get the quad hovering two or three meters from the ground if I put my hand under the sonar, the quad does not react moving up.
Barometer is properly isolated with foam, and sonar and barometer have correct readings in Sensors tab.
I believe vibrations are within the expected values, based on my limited knowledge from inspecting Black Box logs.
Moreover, installing Ardupilot on this same setup with default PIDs proves that there is no problems with my hardware: the altitude hold works wonderfully and the sonar reacts properly.
I’m sharing Black Box log that reflects these two issues.
Every time I moved forward the quad, it dropped altitude until I add some throttle to compensate.
In the last moments of the flight the quad was hovering about three meters from the ground in Position Hold until it slowly moved down and reached the ground, moment which I disarmed the quad.
I’ve spent several hours looking for videos and posts regarding this kind of behavior but I don’t find anyone facing the same problem as I am.
Can someone give me some light where can be the problem and give some hope to move forward?
Thank you.

Tiago
[LOG00002.zip](https://github.com/iNavFlight/inav/files/4076446/LOG00002.zip)

digitalentity, 2020-01-18:
What's your config? Can you post a config dump?

tibean, 2020-01-19:
Hi.
My frame is a 3D printed frame based on Pavel's design (https://www.thingiverse.com/thing:2686018).
Triblade 5045 props, Multistar BLHELI-S 30A ESCs from HK, 3s LIPO battery, EMAX RS2205 motors, sonar TFmini, GPS+mag Beitian BN880.
My configs:
# diff all

# version
# INAV/MATEKF405 2.3.0 Nov 24 2019 / 10:52:41 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gf

# feature
feature -OSD
feature DYNAMIC_FILTERS
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 2 32 115200 115200 0 115200
serial 3 65536 115200 115200 0 115200
serial 4 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 4 1300 2100
aux 1 1 0 900 1700
aux 2 3 0 1300 1700
aux 3 9 0 1800 2100
aux 4 8 1 1800 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_lpf_type = PT1
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set acc_hardware = MPU6500
set acczero_x = -14
set acczero_y = -24
set acczero_z = -36
set accgain_x = 4084
set accgain_y = 4081
set accgain_z = 4100
set rangefinder_hardware = BENEWAKE
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = -46
set magzero_y = -63
set magzero_z = 12
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set motor_pwm_rate = 4000
set motor_pwm_protocol = DSHOT150
set failsafe_procedure = RTH
set vbat_scale = 1099
set current_meter_offset = 30
set yaw_jump_prevention_limit = 200
set model_preview_type = 3
set servo_pwm_rate = 330
set applied_defaults = 2
set gps_provider = UBLOX7
set gps_sbas_mode = AUTO
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_use_midthr_for_althold = ON
set frsky_pitch_roll = ON
set name = J2

# profile
profile 1

set mc_p_pitch = 58
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set use_dterm_fir_filter = OFF
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
set yaw_p_limit = 300
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set battery_capacity = 2200

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

thijn1234, 2020-01-20:
the max altitude for althold is defult 2 meter i have also the tfmini 

pleas do this
inav_max_surface_altitude = 1000
nav_max_terrain_follow_alt = 1000 

this altitude is 10Meter

tibean, 2020-01-20:
Hi, thanks for your reply.

Maybe you mean
inav_max_surface_altitude = 1*0*000 (cm)
nav_max_terrain_follow_alt = 1*0*000(cm)

to get 10meters?

regards,
T

à(s) 07:40:

closed 2020-04-04 by stale[bot]