iNavFlight/inav #5334: Adding NOX Softserial port

tonyyng, 2020-01-18:
I'm trying to add a SoftSerial port to the NOX target to allow SmartAudio VTX control using the LED pad.

I'm a software developer, but don't have experience with this type of hardware. I hope this is the appropriate place to ask these questions.

I looked over the existing code for [NOX ](https://github.com/iNavFlight/inav/tree/master/src/main/target/NOX )and have a couple of questions.

There appears to be a SoftSerial 1 defined already as a workaround for ESC telemetry:
```
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN      PA2 // Workaround for softserial not initializing with only RX
#define SOFTSERIAL_1_RX_PIN      PA2 // Backdoor timer on UART2_TX, used for ESC telemetry
```
If I don't need ESC telemetry, can I redefine SoftSerial1 on PA0 (LED PAD). My current plan is to define SoftSerial2 to PA0.
```
#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_TX_PIN      PA0 // Softserial on the LED pad
#define SOFTSERIAL_2_RX_PIN      PA0 
```

In target.c, there is a timerHardware section I don't understand.
```
const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM2, CH3, PB10, TIM_USE_PPM,  0, 0),   //PPM
    DEF_TIM(TIM2, CH1, PA0,  TIM_USE_LED,  0, 0), //2812LED
    DEF_TIM(TIM5, CH2, PA1,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1_OUT
    DEF_TIM(TIM1, CH1N, PA7,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2_OUT
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT
    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT
    DEF_TIM(TIM9, CH1, PA2,  TIM_USE_ANY,  0, 0), //UART2 TX
    DEF_TIM(TIM9, CH2, PA3,  TIM_USE_ANY,  0, 0), //UART2 RX
};
```
I assume the line `DEF_TIM(TIM2, CH1, PA0, TIM_USE_LED` needs to change, but I'm not sure to what. 
Are TIM2 and CH1 still okay? Looking at other targets, there is a mix of TIM_USE_PPM and TIM_USE_ANY for softserial. 

```
    DEF_TIM(TIM2, CH1, PA0, TIM_USE_ANY,  0, 0), // Softserial on LED PAD
```

Note, I'll make my changes to a new target using the appropriate #ifdefs. I didn't want to complicate the discussion here.

Thanks for any help.

PachisPachis, 2020-02-10:
Hi m8

I just modded and compiled for TARGET=NOX to get softserial working and also I2c. Tested with GPS on softserial on LED PAD and PPM PAD. I2c is on UART1. You can still using UART1 as uart and also LED/PPM as their original function (but not both functions at once).

target.mk
`F411_TARGETS    += $(TARGET)
FEATURES        += VCP ONBOARDFLASH

TARGET_SRC = \
			drivers/accgyro/accgyro_mpu6000.c \
			drivers/barometer/barometer_bmp085.c \
			drivers/barometer/barometer_bmp280.c \
			drivers/barometer/barometer_ms56xx.c \
			drivers/compass/compass_hmc5883l.c \
			drivers/compass/compass_mag3110.c \
			drivers/compass/compass_qmc5883l.c \
			drivers/compass/compass_ist8310.c \
			drivers/compass/compass_lis3mdl.c \
			drivers/pitotmeter_adc.c \
			drivers/light_ws2811strip.c \
            drivers/flash_m25p16.c \
            drivers/max7456.c`

target.h
`/*
 * This file is part of Cleanflight.
 *
 * Cleanflight is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * Cleanflight is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with Cleanflight.  If not, see <http://www.gnu.org/licenses/>.
 */

#pragma once

#define TARGET_BOARD_IDENTIFIER "NOX1"
#define USBD_PRODUCT_STRING  "NoxF4V1"

#define LED0                    PA4

#define BEEPER                  PC13
#define BEEPER_INVERTED

//I2C
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C1_SCL               PB6
#define I2C1_SDA               PB7
#define I2C2_SCL               NONE
#define I2C2_SDA               NONE
#define I2C3_SCL               NONE
#define I2C3_SDA               NONE
#define DEFAULT_I2C_BUS	 BUS_I2C1
#define I2C_DEVICE_1_SHARES_UART1

#define TEMPERATURE_I2C_BUS     BUS_I2C1

#define USE_MAG
#define MAG_I2C_BUS             BUS_I2C1
#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
#define USE_MAG_IST8310
[... 137 lines cut ...]
 */

#include <stdbool.h>
#include <platform.h>

#include "drivers/io.h"

#include "drivers/dma.h"
#include "drivers/timer.h"

const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM2, CH3, PB10, TIM_USE_PPM,  0, 0),   //PPM
    DEF_TIM(TIM2, CH1, PA0,  TIM_USE_LED,  0, 0), //2812LED

    DEF_TIM(TIM5, CH2, PA1,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1_OUT
    DEF_TIM(TIM1, CH1N, PA7,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2_OUT
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT
    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT

    DEF_TIM(TIM9, CH1, PA2,  TIM_USE_ANY,  0, 0), //UART2 TX
    DEF_TIM(TIM9, CH2, PA3,  TIM_USE_ANY,  0, 0), //UART2 RX
};

const int timerHardwareCount = sizeof(timerHardware) / sizeof(timerHardware[0]);
`

PachisPachis, 2020-02-10:
Hi m8

I just modded and compiled for TARGET=NOX to get softserial working and also I2c. Tested with GPS on softserial on LED PAD and PPM PAD. I2c is on UART1. You can still using UART1 as uart and also LED/PPM as their original function (but not both functions at once).

target.mk
```
F411_TARGETS    += $(TARGET)
FEATURES        += VCP ONBOARDFLASH

TARGET_SRC = \
			drivers/accgyro/accgyro_mpu6000.c \
			drivers/barometer/barometer_bmp085.c \
			drivers/barometer/barometer_bmp280.c \
			drivers/barometer/barometer_ms56xx.c \
			drivers/compass/compass_hmc5883l.c \
			drivers/compass/compass_mag3110.c \
			drivers/compass/compass_qmc5883l.c \
			drivers/compass/compass_ist8310.c \
			drivers/compass/compass_lis3mdl.c \
			drivers/pitotmeter_adc.c \
			drivers/light_ws2811strip.c \
            drivers/flash_m25p16.c \
            drivers/max7456.c
```

target.h
```
/*
 * This file is part of Cleanflight.
 *
 * Cleanflight is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * Cleanflight is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with Cleanflight.  If not, see <http://www.gnu.org/licenses/>.
 */

#pragma once

#define TARGET_BOARD_IDENTIFIER "NOX1"
#define USBD_PRODUCT_STRING  "NoxF4V1"

#define LED0                    PA4

#define BEEPER                  PC13
#define BEEPER_INVERTED

//I2C
#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C1_SCL               PB6
#define I2C1_SDA               PB7
#define I2C2_SCL               NONE
#define I2C2_SDA               NONE
#define I2C3_SCL               NONE
#define I2C3_SDA               NONE
#define DEFAULT_I2C_BUS	 BUS_I2C1
#define I2C_DEVICE_1_SHARES_UART1

#define TEMPERATURE_I2C_BUS     BUS_I2C1

#define USE_MAG
#define MAG_I2C_BUS             BUS_I2C1
[... 143 lines cut ...]

#include <stdbool.h>
#include <platform.h>

#include "drivers/io.h"

#include "drivers/dma.h"
#include "drivers/timer.h"

const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM2, CH3, PB10, TIM_USE_PPM,  0, 0),   //PPM
    DEF_TIM(TIM2, CH1, PA0,  TIM_USE_LED,  0, 0), //2812LED

    DEF_TIM(TIM5, CH2, PA1,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1_OUT
    DEF_TIM(TIM1, CH1N, PA7,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2_OUT
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT
    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT

    DEF_TIM(TIM9, CH1, PA2,  TIM_USE_ANY,  0, 0), //UART2 TX
    DEF_TIM(TIM9, CH2, PA3,  TIM_USE_ANY,  0, 0), //UART2 RX
};

const int timerHardwareCount = sizeof(timerHardware) / sizeof(timerHardware[0]);

```

tonyyng, 2020-02-11:
Thanks for your code.  I looked it over to try to understand it.  It is not necessary to change these lines in target.c when you are using softserial on the LED and PPM pads?

    DEF_TIM(TIM2, CH3, PB10, TIM_USE_PPM,  0, 0),   //PPM
    DEF_TIM(TIM2, CH1, PA0,  TIM_USE_LED,  0, 0), //2812LED

I'm planning to use GPS on UART1, Frsky FPORT on UART2 and SmartAudio VTX control on Softserial TX (PPM PAD).  Will that combination work?

PachisPachis, 2020-02-25:
I guess there is some kind of conditional anywhere, you can preserve that lines as is. I have already a working GPS on softserial1, located on pins PPM and LED. Your combination should work. Regards.

closed 2020-05-09 by stale[bot]

Feldsalat, 2020-05-09:
I would also need a soft serial port for SmartAudio, preferably on the LED pad. Could someone maybe add that?

s2kaka, 2020-08-08:
Can Youy show me how to connect Fport, Gps on this board!
I use UART 1 for Fport, UART2 for GPS but GPS not work.
When use Uart 2 for Fport, UART 1 for GPS, Gps work but Fport not work.