iNavFlight/inav #5336: Change OLED Display?

Goatspunk, 2020-01-19:
## Current Behavior
https://github.com/iNavFlight/inav/blob/master/docs/Display.md

> There is currently no way to change the information on the display

## Desired Behavior
I would like a way to change the OLED display.  

Where does the OLED display get its information from?  Can we modify the screen somehow to show more relevant information?  This could be set up as a HUD for a camera (eg. DJI) to show your home arrow, distance to home, alt, speed, mah.

There are all sorts of fun things you could do with this as a HUD.

Neuroplant, 2020-02-08:
Hello,
I prefer a customable OLED too, but I dont understand how you want to use it as a HUD.
Or do you mean the OSD?

closed 2020-04-23 by stale[bot]

buzkill420, 2020-11-27:
Reopen!

Narnasqueneth, 2020-11-28:
Yes please reopen- I've just got my oled screen working and it would be awesome to have system errors displayed (maybe scrolling text if message too long)?

buzkill420, 2020-11-28:
> Yes please reopen- I've just got my oled screen working and it would be awesome to have system errors displayed (maybe scrolling text if message too long)?

Agreed! I know with osd and telemetry/lua its not overly needed these days,  but my lord is it handy out on the field on non-fpv gear, and things without telemetry.  And even fpv its super handy for "at a glance" info.

I know this isn't quite the place, but while we're talking about it, can anyone point me in the right location as to where it describes in detail, what is displayed on screen... not quite sure on at least one or two things. 

Thanks

Narnasqueneth, 2020-11-28:
> > Yes please reopen- I've just got my oled screen working and it would be awesome to have system errors displayed (maybe scrolling text if message too long)?
> 
> Agreed! Or maybe the option to have custom scrolling text/ image... idk if that's cool or not. I think it would be neat. . And i know with osd and telemetry/lua its not overly needed these days, but my lord is it handy out on the field on non-fpv gear, and things without telemetry. And even fpv its super handy for "at a glance" info.
> 
> I know this isn't quite the place, but while we're talking about it, can anyone point me in the right location as to where it describes in detail, what is displayed on screen... not quite sure on at least one or two things.
> 
> Thanks

I would also love to know what everything means- most of it is straightforward but there are a couple of elements that I'm not sure about (eg "R" in the top right and rotating symbol)

buzkill420, 2020-12-11:
> I would also love to know what everything means- most of it is straightforward but there are a couple of elements that I'm not sure about (eg "R" in the top right and rotating symbol)

Or the "Alt:" value.....i would assume it was altitude but the value does not reflect that...... really wish there was away to customize displayed info, or at least someone pointed out where to change the code in the source

buzkill420, 2021-01-29:
Or how about Arming errors!  >~<    its soooo annoying to not know why your plane is not arming for you!  Heaven forbid you put inav in something with no camera