iNavFlight/inav #5345: DISABLE NAVIGATION FIRST

Sufy561, 2020-01-23:
hello , I am contacting you because I am having an issue with the inav plane that I've recently built: with everything as it should be , I used to be able to arm and fly . 1 week ago my F765 decided not to arm , it had 12 satellites , and everything was as it used to be , I hadn't changed a single thing since I last flew it , but a message popped up on the osd which said: DISABLE NAVIGATION FIRST - UNABLE TO ARM . I have no idea on what could be going on .
This issue popped out of nowhere , and I had never seen it before . Now I can't fly anymore . 
Thank you 
Sufyan

giacomo892, 2020-01-23:
https://lmgtfy.com/?q=cannot+arm+inav

closed 2020-01-23 by giacomo892

alhimik1986, 2020-09-27:
It means that you cannot to arm using NAV_POSHOLD mode. You must switch off it first, then arming, take off and then switch on the NAV_POSHOLD mode.

frangarma, 2020-11-30:
> It means that you cannot to arm using NAV_POSHOLD mode. You must switch off it first, then arming, take off and then switch on the NAV_POSHOLD mode.

Hi, by the way, I had the same problem taht Sufy561, and following your advice _"You must switch off it first, then arming, take off and then switch on the NAV_POSHOLD mode." I can see in OSD that pass from ANGL mode to AIR mode, I mean, I switching in ARM mode, if I up the throttle unitil 1400 (more or less that is empiric) and I switching the switch mode to NAV POSHOLD, I can see into OSD that pass from ARM to AIR mode (note that all this testing has been made in my desktpo not flying), that is correct? Shouldn't I see the NAV POSHOLD into OSD? 
Thanks for your help

gubernatorAtik, 2022-07-15:
I had the same problem. This was visible because I left the RHT key active on the remote and I disabled it. . All is good now

LAFlyer, 2023-04-06:
I have this same problem.  
is RHT key return to home?
My Arm and Return To Home are opposite positions of same switch.
Up is Arm. Mid is Disarm and Down is RTH.  All show correct in Modes page of Configurator.

MrD-RC, 2023-04-07:
> I have this same problem.
> is RHT key return to home?
> My Arm and Return To Home are opposite positions of same switch.
> Up is Arm. Mid is Disarm and Down is RTH. All show correct in Modes page of Configurator.

That is a bad switch setup. Have arm on its own 2 way switch. The model still needs to be armed for RTH to work. 

This is not an issue. This is working as intended.

LAFlyer, 2023-04-07:
Thanks Pawel, but I discovered while I was getting the "Disable Navigation First" warning that RTH activates motor arming.  I had not saved the mode setup when I was testing.  I borrowed the switch setup from Painless360.  Thanks Lee. 
All is well now.

Szczery-FPV, 2023-12-31:
The only solution that comes to my mind is...
If you use the RC ELRS link, change the speed in "Hz" to a lower one, i.e. from 250Hz down, otherwise you only have 8 channels.
You can use speeds from 50 to 250Hz.