iNavFlight/inav #5347: Fast landing on the long distance multirotor

jaguraphg, 2020-01-25:
Hi guys, this is probably more the question, not the bug. I really cannot find the answer somewhere else. I built the long range 8" quadcopter, and the issue is that every time when I try to land from the high altitude - I always have to very carefuly find the position of the stick when the props are still spinning. If I fully move the throttle to the lowest position then it seems that the props are stopped in the air, and after that I never tried to wait for longed that 0.5 sec, I move the stick a little bit up and then the props are stating very very sharply with the loud sound. I fear to drop the stick. So then I cannot control the vertical speed of the landing and cannot set the limit.

ps. Airmode are permanently off because I think that the copter will jump on landing. Not sure if this is correct.

What I am doing wrong? Thank you!

giacomo892, 2020-01-25:
With AIRMODE off and MOTOR_STOP enabled no way to get stabilization in the air.
Keep AIRMODE on and just land softly and promptly disarm.

closed 2020-01-25 by giacomo892

jaguraphg, 2020-01-25:
Got it, thanks. Just as a theoretic question, why could you do the air stabilization in the air without the airmode? The hardware always gives you enough information that you're in the air, the altitude (baro) is being changed, and in this mode why wouldn't just stabilize the multirotor and land with the controlled speed? Thanks.