iNavFlight/inav #5349: Airspeed

david9981, 2020-01-26:
## Current Behavior
I am a competition cross-country sailplane/paraglider pilot, and a full time pilot. One of the most important skills to maximum distance flown, besides knowing where to find thermals, is to always fly at the most efficient airspeed.  From an aerodynamic standpoint, all airplanes have a published best_range and best_endurance speed. Currently, iNav only consider cruise_throttle, and auto-throttle operates between min_throttle and max_throttle to keep the plane in level flight.  We cannot control what airspeed we want to fly at.  Setting the cruise_throttle too low and you might even risk stalling the plane if auto throttle don't kick in fast enough.  iNav uses Airspeed for display purposes only.

## Desired Behavior
ArduPlane has been using airspeed for flight control purposes for quite sometime, including the ability to set min_airspeed as a safety feature to prevent stalls.  The most desired behavior would be to not only have the ability to set target airspeed=best range speed under no-wind situation, but also the ability to auto adjust the best_range speed according to actual wind conditions. 

## Suggested Solution
Utilize the airspeed parameter taken from airspeed sensor, and incorporate it into an auto-throttle mode during cruise.  In the future, similar to Arduplane, this parameter can be used in a variety of scenarios including stall prevention, determining max_climb, max_descend, min_sink (instead of using pitch angle), approach, landing flare height.

## Who does this impact? Who is this for?
Airplane users who want to maximize distance flown.

dsaboot, 2020-01-26:
I really want this feature, now I use the stick to achieve the speed I need on a cruise(nav_fw_allow_manual_thr_increase = ON), but I would like to set the speed in km/h.

Jetrell, 2020-01-27:
@david9981 I use this same sort of function with the Vector flight controller. Its called `closed loop speed mode`.
This functionality is far from plug N play and requires P I loop throttle/airspeed tuning, according to the models weight/drag and thrust and within the systems **tight** boundaries. 
It works well in the Vector, provided its setup correctly(no short cuts). But its not trustworthy in every condition. 
There are too many variables to bring up here. Needless to say, it relies totally on the pitot, which can get clogged etc etc.

I've also seen the arduplane system in action. It too has many limitations.

closed 2020-04-11 by stale[bot]