iNavFlight/inav #5350: Simple and super-simple modes in INAV

dimon777, 2020-01-27:
## Current Behavior

[Simple and super-simple](https://ardupilot.org/copter/docs/simpleandsuper-simple-modes.html) modes in INAV are not available currently.

## Desired Behavior

Implement these modes.

## Suggested Solution

Learn from ardupilot.

## Who does this impact? Who is this for?

These modes are for beginners and/or those who are budget limited to learn the hobby "with the wallet" i.e. ready to fix all crashes with costly repairs. Also for small copters it is really challenging to fly them beyond certain distance (100+ meters) safely as orientation of the drone is difficult to recognize from the ground.

## Additional context

DzikuVx, 2020-01-27:
I think is is an equivalent of a `HEADFREE` mode?

dimon777, 2020-01-28:
I think so. The main difference in super simple the drone moves in circle around the pilot when he applies roll controls regardless of where it is heading.

robertha64, 2020-04-15:
Existing Inav HEADFREE is 100% equivalent of Ardupilot Simple mode (I tested both). In this mode you can return back from one direction if loose orientation, but you should not fly over staring point as then will continue to fly away. 
The Super simple is different as it uses GPS and stored start position and changed behavior when machine fly over pilot head.

closed 2020-12-11 by DzikuVx