iNavFlight/inav #5358: Blackbox writes some trash sometimes

jaguraphg, 2020-01-28:
## Current Behavior
FC Holybro F7 Kakute
I use blackbox, the loop rate is 2kHz, blackbox ratio is 1/2. I have to fly 4-5 times in order to get the accurate logs from the blackbox because about 4 of 5 times it writes some weird stuff to the file with a lot of peaks in all channels. I am attaching the file from today's flight for example. I open the file in the Explorer and it shows that the copter just rotated in air entire time. At the same time about each 5th flights is recorded correctly.

I tried to reduce the ratio even more, also tried to change the micrSD card. The same.

## Steps to Reproduce
1. Enable blackbox
2. Insert SD card
3. Fly
4.

# INAV/KAKUTEF7 2.3.0 Nov 24 2019 / 10:29:29 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

[LOG00047.TXT](https://github.com/iNavFlight/inav/files/4124218/LOG00047.TXT)

b14ckyy, 2020-02-02:
saw some logs on my Talon GT SD card too with exact the same behavior. Unfortunately I did not safe them as I thought it may be an SD failure due to moisture that day high up in the mountains. 

FC is Matek F411SE

jaguraphg, 2020-02-10:
I have a lot of such logs. At last time it always writes this trash to the card. I changed a couple of SD cards, not sure if the problem is in the card itself.

jaguraphg, 2020-02-10:
Tried to flash 2.4 today, will try to fly tomorrow. But I do not think something was changed in this version.

closed 2020-04-25 by stale[bot]