iNavFlight/inav #5362: New flight regulations requires Cylindrical Geofence

rsilk1949, 2020-01-30:
## Current Behavior
There is currently no way in iNav 2.3 to limit altitude and distance from home which in many countries is now a mandatory regulation.

## Desired Behavior
Match cylindrical Geofence functionality  of Ardupilot.
More complex polygon geofencing is not required to comply with new regulations.

## Suggested Solution
New parameters to set maximum altitude above home and maximum radius from home.
Action options:  RTH or Land if fence exceeded.
Would apply to all flight modes if GPS is available.

## Who does this impact? Who is this for?
All users who are subject to new regulations imposed by FAA, CAA, EASA, etc.
Could also be useful to others who might lose control or visual contact.

digitalentity, 2020-01-30:
INAV as of now doesn't enforce regulatory compliance - it's the pilot's responsibility to respect local rules

rsilk1949, 2020-01-30:
This feature will assist the pilot to comply with the local rules. It should be implemented as an optional feature.

giacomo892, 2020-01-30:
... as much any car manufacturer automatically software restrict speed on the cars they sell

rsilk1949, 2020-01-30:
That is called Cruise Control :)

stronnag, 2020-01-30:
Duplicates 

https://github.com/iNavFlight/inav/issues/62

https://github.com/iNavFlight/inav/pull/4798

closed 2020-01-30 by digitalentity