iNavFlight/inav #5372: VTX settings are not written to VTX when using a power switch

jj-uk, 2020-02-01:
iNav 2.3.0
Matek F722-SE Wing with "vsw" power switching via "USER 1" modes.
Matek VTX-HV (http://www.mateksys.com/?portfolio=vtx-hv#tab-id-5)

I have my VTX on a power switch (USER 1) so that the VTX is not powered when the flight controller boots.

It appears the iNav writes the VTX freq, power and channel information to the VTX when the flight controller boots, and not ever again.
As a result, the VTX settings are never written to the VTX. This is my observation.

A solution would be:

USER 1 not configured:
Write the VTX settings afeter flight controller boot.

USER 1 configured:
When USER 1 changes from ACTIVE to NON-ACTIVE (e.g. changes from blue to grey in the configurator modes), wait 2-3 secs for the vtx to power up then write settings to the vtx.

b14ckyy, 2020-02-13:
As a workaround you could use the "Low power on Disarm" Feature in configuration tab and set t to FIRST_ARM or ALWAYS. In this case the the Smart Audio Control Package is sent to the VTX again when you arm the plane to set the power level to the desired one. 

Why not using this feature in general? This will keep the VTX at a low 25mW power level when disarmed and waiting for GPS and power up to your desired power level on arm?

jj-uk, 2020-04-14:
This remains an issue.

closed 2020-06-28 by stale[bot]