iNavFlight/inav #5377: Target for TMOTOR F7 HD?

angeldimitrov, 2020-02-03:
## Current Behavior
No support for TMOTOR F7 HD

## Desired Behavior
Target for TMOTOR F7 HD

## Additional context
Are there any plans to support this FC?

michifpv, 2020-06-07:
Also for the TMOTORF4 HD flight controller

digitalentity, 2020-06-07:
https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

seanreynoldscs, 2020-07-09:
I would also like this. I'm new to FPV and I bought the DJI stack to build out a quad. 
https://store.dji.com/product/t-motor-f55a-pro-f4-hd-stack

It works great in Betaflight with around 4% cpu load, but I would like to add a few sensors for some extra automation like hover in place since I'm just starting out. 

Are there plans to add the firmware for the motorf4 HD for iNav? It should be a little easier since it exists for Betaflight already.

seanreynoldscs, 2020-07-09:
I just checked the requirements: 
Check: On-board sensors include at least the IMU (gyroscope + accelerometer)
Check: At least 2 hardware serial ports are available with both TX and RX pins
Check: At least 512K of firmware flash memory and at least of 64K of RAM available
Check: At least one I2C bus broken out (SCL and SDA pins) and not shared with other functions

t-motor f4 hd  has ICM20602 
t-motor f4 hd has 4 Tx and Tx
STM32F405RGT6 has 1024 kB program memory and 192kB of ram.
t-motor f4 has i2c exposed and is not shared with other functions. 

Also this has Firmware: BetaFlight Target: TMOTOR F4
available as a starting point.

nyartboy, 2020-09-04:
Throwing my hat into this ring of requests. Would love a target for the T-Motor F7HD. Thanks!

fpetras, 2020-09-14:
Would love to try out INAV with my T-Motor F7 as well!

johnnyo21, 2020-10-05:
+1 for the t-motor F7 -- Would love to have some of the iNav features.

Vanvaulder, 2021-05-02:
Target  T-Motor F7 HD pro ll 
Can we please get this on INAV?

Sedgee, 2021-05-03:
I have posted elsewhere requesting this and would just like to add my vote to getting T-Motor F7 HD target.

jimsynz, 2021-05-21:
I have a build which seems to be working for me.  Is anyone willing to alpha test a build for me?

nyartboy, 2021-05-21:
Darn, I don’t have a T-Motor in service at the moment. I have a Diatone mamba build that is ready tho. Ultralight 5” with GPS and crossfire

famascl3m, 2021-06-13:
> I have a build which seems to be working for me. Is anyone willing to alpha test a build for me?

Hi, i have a fresh 7 inch build with tmotor f7 hd 55a stack and i would try your alpha version if you are ok ! 😁

jimsynz, 2021-06-13:
[here](https://www.dropbox.com/sh/qmrndccnj85mqgz/AAAVH4vRxFEtcjiXb7uhwHiUa?dl=0) are builds I've just made against master at 99c06d6a93e8e0e9f10476fe309fa03473c03aa8. Please be very careful with this build!

famascl3m, 2021-06-24:
It seem to have a problem with the esc communication, i have setup the calibration, the mixer and then in the output tab with dshot 600 the esc2 doesnt spin the motor, if i reflash betaflight the 4 esc work instantanly (only with bf), BLHeliSuite with inav says ESC2 doesnt have the same input protocol. I have tried multiple reflash with full chip erase without better result

[image]

jimsynz, 2021-06-24:
Thanks for that. I'll double check the target configuration this weekend.

iggut, 2021-08-17:
Any chance this could be made to work with the non HD Tmotor F722?
Please.

ojmarek, 2021-10-16:
i dream of a software on tmotor velox f7

jimsynz, 2021-10-17:
I have sold my T-Motor F7 flight controller, so won't be doing any more work on this branch.  Good luck!

closed 2022-01-26 by DzikuVx

loic-png, 2023-03-30:
I recently bought the t-motor f7 hd flight controller.
Is it possible to use INAV 6 with it?
Or can I only use Betaflight?
Thank u.

S3de11, 2024-06-19:
Does it support tmotor f7 hd ll?

sensei-hacker, 2026-01-07:
> I flashed the TMotor F7 with iNAV 8xx and 9xx. It seems okay but if i want to calibrate the ACC it is not possible. GPS is working well. Has someone Ideas about the F7 TMotor? I find the position hold and altitude function in iNAV better than in Betaflight.

Can you tell us *exactly* what problem you are having when you try to calibrate the accelerometer?
You should click the calibrate button, then with the board flat, click it again. Then with the board upside-down, click it again.

Note if the board is currently wrapped in an aircraft it can be more difficult, as you need to get the board flat with the earth.