iNavFlight/inav #5378: MAMBA F722S problem dshot300 and dshot600

bizio996, 2020-02-03:
I too would like to use mamba f722s with inav, but the protocols esc 300 and 600 works really badly, now it work with multishot. Please you see the video on test dshot 300 and dshot 600 don't work, when I get up throttle one motor is stopped is this happened in the fly!
i also tried with betaflight and it all works with my esc dshot 300 and 600
PS. I'm use last version inav 2.4
https://youtu.be/9NiGlqb5OEA  (test motor spin)

here you can see some flight tests with multishot protocol

https://youtu.be/NI2bVEjLd-A (1° test multishot inav2.4 pid stock)
https://youtu.be/v0TwZOdUE_4 (2° test multishot inav2.4)
https://youtu.be/iA1zj6HkVEk (3° test multishot inav2.4)
https://youtu.be/RuWK8nTHk3Y (4° test multishot inav2.4)

digitalentity, 2020-02-05:
Verified with the logic probe.
On MAMBAF722 target DSHOT output timings are up to spec and identical on Betaflight and INAV.

Specifically for DSHOT600:
INAV 2.4-RC2: 583ns / 1167ns / 600kHz
BF 4.1.1: 583ns / 1167ns / 600kHz

Left the board with INAV running for 1 hour - no issues with signal breakage.

Questions:
1) Please provide a full config dump for the issue
2) At what throttle do the motor stop? At 100% or lower?
3) Is it always one motor or different motors?
4) Does DSHOT150 work or all digital protocols break?

digitalentity, 2020-02-05:
@bizio996

bizio996, 2020-02-07:
hi, thanks for your support, today I also tried with dshot 150 and when I turn up the gas the engine stops and the esc is restarted.  I don't really know what it can be.

bizio996, 2020-02-07:
I tried from the inav configurator, if I pull up the master all the motors run, while if I pull on the single motors when I do it with the motor 4 it does not turn and my esc restarts, I did the same test with bf and everything works, it does not I really understand the reason. what firmware version did you use for your test? I currently have the unstable fw version nstalled on the my flight controller. Thanks again.
I attach the dump.
[dump_mamba_f722.txt](https://github.com/iNavFlight/inav/files/4172345/dump_mamba_f722.txt)

digitalentity, 2020-02-08:
Ok, bug confirmed. Looks like F722 doesn't like (DMA2 Stream6 Channel0) combination. Can't spot any obvious bugs in DMA/Timer code, so will simply move the M3 to a different DMA

digitalentity, 2020-02-08:
Interestingly, F4 doesn't have this issue, so it's F7-only

digitalentity, 2020-02-08:
Should be fixed now. Closing

closed 2020-02-08 by digitalentity

Jacky2k, 2020-02-09:
I can confirm that this fixes the dshot issue on my MAMBAF405 :)
Thanks a lot!

bizio996, 2020-02-10:
> Should be fixed now. Closing

Hi, thanks for your interest and support, how do I install the updated firmware on my fc? from the configurator I see the firmware only if I enable the display of unstable versions.

bizio996, 2020-02-10:
> Ok, bug confermato. Sembra che F722 non ami la combinazione (DMA2 Stream6 Channel0). Non è possibile individuare eventuali bug evidenti nel codice DMA / Timer, quindi sarà sufficiente spostare M3 su un DMA diverso

ok, so I have to remap the motor 3 resource on another pad, do I understand correctly?

giacomo892, 2020-02-10:
@bizio996 you don't have to do anything. Just wait for final 2.4

Jacky2k, 2020-02-10:
If you want the patch right now and cannot or don't know how to build the hex file, let us know. Someone will probably do it for you :)

bizio996, 2020-02-10:
> Se vuoi la patch in questo momento e non puoi o non sai come creare il file hex, faccelo sapere. Qualcuno probabilmente lo farà per te :)

if I could have the hex file to try it would be nice!

Jacky2k, 2020-02-10:
What is the target name?

bizio996, 2020-02-11:
mamba f722s

bizio996, 2020-02-11:
> Qual è il nome target?

mamba f722s

Jacky2k, 2020-02-11:
[inav_2.4.0_MAMBAF722.zip](https://github.com/iNavFlight/inav/files/4184501/inav_2.4.0_MAMBAF722.zip)
This is current development branch.