iNavFlight/inav #5379: F411 wing incorrect OSD Altitude

egpurush, 2020-02-03:
I have a Bixler 3 with F411 wing with M8N gps  module, The OSD displays incorrect altitude reading, every time you power the  flight controller, some time the value the 0, and negative value and high altitude while it is on ground. Some the value dont reduce when the altitude is couple of meter abuve the ground, not sure the cause of the problem.  

Please find video link https://youtu.be/AArpPa0AqHc (Ignore the RSSI reading value)

I have Inav 2.3.0 version and below is the diff of the configuration.

# version
# INAV/MATEKF411_SFTSRL2 2.3.0 Nov 24 2019 / 11:03:04 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 1 100 0 -1
smix 1 4 0 100 0 -1
smix 2 5 0 100 0 -1
smix 3 6 2 100 0 -1

# servo
servo 3 1000 2000 1400 100
servo 4 1000 2000 1417 -100
servo 5 1000 2000 1550 -100
servo 6 1000 2000 1550 100

# logic

# gf

# feature
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 2 115200 115200 0 115200
serial 30 2048 115200 115200 0 115200
serial 31 32 115200 115200 0 115200

# aux
aux 0 0 0 1500 2100
aux 1 1 1 1300 1700
aux 2 2 1 1700 2100
aux 3 4 4 1700 2100
aux 4 3 4 1300 2100
aux 5 8 9 1700 2100
aux 6 10 1 900 1300
aux 7 11 5 1700 2100

# adjrange
adjrange 0 3 0 900 2100 49 10

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 2 2 V
osd_layout 0 1 25 1 V
osd_layout 0 2 0 0 V
osd_layout 0 5 23 10 V
osd_layout 0 6 23 9 V
osd_layout 0 7 14 12 V
osd_layout 0 9 0 3 H
osd_layout 0 10 0 10 V
osd_layout 0 11 25 3 V
osd_layout 0 12 24 4 V
osd_layout 0 13 22 6 V
osd_layout 0 14 1 1 V
osd_layout 0 15 3 6 V
osd_layout 0 20 20 12 V
osd_layout 0 21 1 12 V
osd_layout 0 22 14 1 V
osd_layout 0 23 12 2 V
osd_layout 0 28 24 9 H
osd_layout 0 30 2 11 V
osd_layout 0 32 25 2 V
osd_layout 0 33 0 3 V
osd_layout 0 40 1 8 V
osd_layout 0 45 0 0 V
osd_layout 0 96 3 7 V
osd_layout 0 104 13 5 H
osd_layout 0 106 13 2 H

# master
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6000
set acczero_x = 27
set acczero_y = -7
set acczero_z = -113
set accgain_x = 4085
set accgain_y = 4114
set accgain_z = 4063
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set failsafe_procedure = RTH
set yaw_jump_prevention_limit = 200
set platform_type = AIRPLANE
set model_preview_type = 14
set small_angle = 180
set applied_defaults = 3
set gps_sbas_mode = AUTO
set nav_rth_allow_landing = NEVER
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 5000
set osd_video_system = NTSC
set osd_alt_alarm = 300
set name = Bixler-3 
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5843

# profile
profile 1

set fw_p_pitch = 20
set fw_i_pitch = 30
set fw_ff_pitch = 15
set fw_p_roll = 20
set fw_i_roll = 30
set fw_ff_roll = 15
set fw_p_yaw = 45
set fw_i_yaw = 5
set fw_ff_yaw = 15
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set pitch_rate = 15
set yaw_rate = 9

# battery_profile
battery_profile 1

set bat_cells = 4
set battery_capacity = 6000
set battery_capacity_warning = 3000
set battery_capacity_critical = 1800

# end the command batch
batch end

egpurush, 2020-02-07:
Can someone look on the issue and help me to get it solver

egpurush, 2020-03-02:
Waiting for some help....

stronnag, 2020-03-02:
This is not a help forum, you may have a long wait.

closed 2020-05-15 by stale[bot]