iNavFlight/inav #5381: MATEKSYS VTX not working on softserial with F411 wing (IRC Tramp protocol)

ShikOfTheRa, 2020-02-04:
## Current Behavior
Hello

I have a MatekSys F411 wing and MatekSys VTX 

If I connect VTX control to hardware TX port it works perfect.
If I connect VTX control to softserial it does not work.

I have also used a build with 2 * soft serials to try and alternative pin. Again with no luck

I have googled many sites and found several with same issue but no one with it working. The ones that resolve it are able to do so by moving to a hardware port.

I am running fport serial rx and also GPS, so both hardware ports are tied up. But it fails also with a basic build and just softserial enabled.

TBS Smart audio seems to work on soft serial. Issue appears to be with IRC tramp protocol

## Steps to Reproduce

1. Connect MatekSys VTX (IRC tramp protocol) to hardware serial
2. Verify it works
3. Reconfigure and change to soft serial
4.Verify it doesn't work

## Additional context

Dump:
https://pastebin.com/5PUXpvbW

Board: MATEKSYS F411 WING
VTX: MATEKSYS VTX-HV

# INAV/MATEKF411 2.3.0 Nov 24 2019 / 11:01:21 (d0bdd38c0)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

b14ckyy, 2020-02-08:
Do you mean the VTX-HV? http://www.mateksys.com/?portfolio=vtx-hv
Works fine for me with F411 Wing and Tramp over soft serial.

ShikOfTheRa, 2020-02-09:
Thanks for the feedback - that's appreciated. 
TBH, I was sure something wrong with my setup, but it works fine with a hardware pin and I'e dried two different soft serial pins. Then googled and could find anyone with a working setup.

Any chance you could put a dump of your settings into pastebin?

b14ckyy, 2020-02-09:
[INAV_cli_Kelly_2.3.0_20191126_174128.txt](https://github.com/iNavFlight/inav/files/4178144/INAV_cli_Kelly_2.3.0_20191126_174128.txt)

here you go :)

ShikOfTheRa, 2020-02-10:
Aesome. Thanks a lot for your time. I'll load this up and test tomorrow.

b14ckyy, 2020-02-10:
> Aesome. Thanks a lot for your time. I'll load this up and test tomorrow.

I just checked your dump too. The serial mapping is exact the same as mine. Can you post a picture how the cabling is done? So at what pins on the VTX and FC the control line is connected?

ShikOfTheRa, 2020-02-10:
I pasted your dump into my f411 – still no luck.

Setup is as per the inav recommended table for FPORT/GPS and VTX control
GPS on uart2 tx/r (works ok)
FPORT on UART1 rx (works ok)
Vtx tx pin to FC ST1 pin 
I also flashed the 411 build with 2 soft serial ports – tried both soft serial (st1 and led pin) with no luck
If I disabled GPS and connected Vtx tx to uart2 tx it works fine.
I also flashed RSSi build to test analog on ST1 – works fine.

I have another board hopefully arriving tomorrow to try. FC seems OK, but could always be a minor part of the processor. Unlikely but right now I’ll try anything. 
Just wanted define not working…
OSD is set to display channel / power. This works fine with hardware uart.
I made assumption that it works same with soft serial, but I am wondering this now.

Can’t easily take photo as it’s all embedded in nano goblin, but I’ll pull out and do that if can’t solve issues.
Sure wiring is OK as if I swap th evtx control wire between soft/uart it works.

And I just found this for VTX firmware. I will have to test…
V1.4 （Sep.24, 2017）
Fixed BFCMS control on softserial-tx

jbman, 2020-02-10:
I have the same board and Firmware version INAV/MATEKF411 2.3.0
and connected a "FuriousFPV VTX Stealth Nano Race" which claims to support IRC Tramp functionality. It is also connected on Softserial ST1, but settings in Configurator or via OSD have no effect.

ShikOfTheRa, 2020-02-11:
Just knowing this works for b14ckyy helps a lot. Tonight I will test with vtx firmware update and FC (if it arrives). I'll let you know how it goes.

ShikOfTheRa, 2020-02-11:
I replaced the FC and problem fixed.
A little puzzled because also tried soft serial on led pin and tested the pin as analogue RSSI. Weird.
 
@b14ckyy  - thanks for the help. Just knowing it worked for someone and a good config helped because all I could see was posts about it not working.

ShikOfTheRa, 2020-02-12:
Bit more extra testing last night. Interestingly since the VTX firmware update, softserial 2 now works on ledpin so I think the VTX must have have old firmware on it as well.

1 - Mateksys VTX-HV needs to run v1.4 firmware to run softserial
2 - Replace FC board - faulty hardware pin

closed 2020-02-12 by ShikOfTheRa

fpvMarkS, 2024-03-27:
I had this same issue with VTX control over softserial with the Lumineer TX5GS ... Updated the firmware on the VTX following the instructions on GET FPV website that linked to the Matek site ... here ... 
https://www.mateksys.com/?portfolio=vtx-hv#tab-id-6