iNavFlight/inav #5384: Optical Flow/Lidar sensor not working (Matek 3901-L0X)

hydra, 2020-02-04:
## Current Behavior
Optical flow and Sonar sensor indicators remain red in UI.

## Steps to Reproduce
1. Flash SPRacingAIRBIT firmware, built from https://github.com/spracing/inav/tree/spracing-airbit
2. Connect Matek optical flow sensor unit 3901 to UART3
3. Enable MSP on UART3.
4. Choose `MSP` for Sonar and Optical Flow sensors.

## Expected behavior
Blue icons indicating that optical flow/sonar sensor is working.

## Additional context
* Also tested with RX/TX pins swapped, just in case.
* You can use SPRACINGF3EVO FC hardware to test with, the SPRACINGAIRBIT is a variant of the SPRACINGF3EVO but has OPTICAL FLOW support compiled in.  MCU pins for UARTS, gyros, i2c, etc are the same.

- FC Board name and vendor: SPRACINGAIRBIT

Diff/Dump attached.

[inav-opflow-spracingairbit.zip](https://github.com/iNavFlight/inav/files/4155794/inav-opflow-spracingairbit.zip)

Firmware attached, which can be safely flashed to SPRacingF3EVO - built from tree linked above.

[inav_2.3.0_SPRACINGAIRBIT.zip](https://github.com/iNavFlight/inav/files/4155806/inav_2.3.0_SPRACINGAIRBIT.zip)

Note: I don't have sensor hardware to test, but can get the user to perform tests and I can build new binaries as appropriate.

Note: GCC 8.3.1 used to compile as 8.2 is broken on windows/cygwin.

hydra, 2020-02-04:
More context:

Sensor:
http://www.mateksys.com/?portfolio=3901-l0x

Setup-guide:
http://www.mateksys.com/downloads/setup_optic%20flow_in_INAV_rev1.pdf

Target.h change to enable optical flow/rangefinder via MSP:
https://github.com/spracing/inav/blob/spracing-airbit/src/main/target/SPRACINGF3EVO/target.h#L184-L189

Relevant lines from diff:
```
serial 2 1 115200 115200 0 115200
set rangefinder_hardware = MSP
set opflow_hardware = MSP
```

digitalentity, 2020-02-05:
@hydra optic flow and rangefinder are only supported on targets having more than 256K of flash. F3 boards don't have enough flash memory for those cool features.

Also, it would be good to have the target supported by mainline INAV first, we can't afford wasting resources on bugs filed against forked code.

hydra, 2020-02-05:
@digitalentity I'll do a PR for the target, I did note that sensors are normally only for targets with more than 256K flash, but I did enable the features specifically in the target file after disabling other features that will never be used on the target.

hydra, 2020-02-05:
@digitalentity See https://github.com/iNavFlight/inav/pull/5387

digitalentity, 2020-02-05:
Sorry,  didn't notice that you enabled the features in the target file.
In this case you may want to verify with a logic probe that the optic flow board is actually sending data.

hydra, 2020-02-05:
> Sorry, didn't notice that you enabled the features in the target file.

That's OK!

> In this case you may want to verify with a logic probe that the optic flow board is actually sending data.

Yes I have requested that @makeadrone does that, waiting for his results.

Can you think of anything else to check?

Also, as an-aside, it's probably fairly useful, long-term, to get the SPRacingAIRBIT working with the OpFlow/Lidar sensors as it's an educational target and the iNAV support is targeted at the advanced students and who knows what their inquisitive young minds might bring to us in the future!

closed 2020-04-19 by stale[bot]