iNavFlight/inav #5390: Motor Stops in Nav mode at minimum throttle

breadoven, 2020-02-07:
Bit confusing looking at the various info on this so perhaps it's easier to ask the simple question:

On a fixed wing should the motor stop when flying in 3D Cruise mode when the throttle stick is reduced to zero ? And if so should the throttle revert to Manual from Auto in the OSD when this happens ?

Baring in mind the following settings are used:

nav_fw_allow_manual_thr_increase=OFF
nav_overrides_motor_stop = ON

Jetrell, 2020-02-07:
I've experienced the same issue.
I also have - `nav_overrides_motor_stop = ON `(default)  and `nav_fw_allow_manual_thr_increase = ON`

**Wiki states** : `nav_overrides_motor_stop = ON.... Setting to OFF combined with MOTOR_STOP feature will allow user to stop motor when in autonomous modes. On most places this setting is likely to cause a stall.`

But the motor still can be shut down in `Cruise mode` while `nav_overrides_motor_stop = ON`

I don't really have an problem with this. Although I do have an audio warning set on my radio Tx, in `AH` and `3CRS`modes, to tell my if I accidentally pull the throttle back in flight, which I have done before and shut the motor off.

breadoven, 2020-02-08:
Would be useful to get clarification of what is supposed to happen. The reason I noticed this during a flight is because at the moment the throttle was reduced to minimum and the motor stopped I was holding the pitch stick full nose down to make the plane descend. When the motor stopped, the OSD throttle went from Auto to Manual and the plane pitched down abruptly as if under direct manual control. I released the pitch stick and it recovered level. It was only after looking at the OSD video after to see why it pitched I realised the motor had stopped ... which I thought it wouldn't do in Nav modes with Auto throttle. Checking the Log again I can see the motor stopped when the throttle stick went below 1100 (rcData[3]), rcCommand[throttle] went to 1127 at this point, stopping the motor. It restarted in Auto throttle when rcData(3) went back above 1100.

Looking at another Log file for a different flight I found a transition from Manual mode to 3D Cruise with the throttle stick at 1150 which meant the motor was off (starts at 1200 with my ESC) in Manual but started when 3D Cruise was selected by switching Manual off. Auto throttle took control.

Looking at my settings 1100 is min_check:

set min_check = 1100
set max_check = 1900
set max_throttle = 1850
set min_command = 1000
set throttle_idle =  15.000
fw_min_throttle_down_pitch = 0

Seems the 1127 value comes from the new set_throttle_idle = (1850-1000)*0.15 = 1127.5

Isn't really a problem, just leave the throttle stick above minimum. However the abrupt pitch down when the motor stopped doesn't seem right since it definitely acted as if it had reverted to direct manual stick control rather than Nav control. Bug ?

breadoven, 2020-02-11:
From what I could glean from the code it seems` nav_overrides_motor_stop` only affects RTH and WP modes. So keep the throttle above `min_check` to avoid problems in the other Nav modes.

closed 2020-02-11 by breadoven