iNavFlight/inav #5397: Matek F405-CTR + Quad + Servo S6

DangerD256, 2020-02-10:
Trying to make at least one working servo for camera Tilt.

[image]

The problem is that there's no PWM signal on S6, tested also with PWM tester.
With arducopter firmware both S5 S6 works fine for gimbal

DzikuVx, 2020-02-19:
INAV on this board allows only to use one servo on S6 in multirotor configuration and only when target MATEKF405_SERVOS6 is used. You are trying to assign two servos. S5 is reserved to LED strips

closed 2020-02-19 by DzikuVx

DangerD1024, 2020-02-19:
> INAV on this board allows only to use one servo on S6 in multirotor configuration and only when target MATEKF405_SERVOS6 is used. You are trying to assign two servos. S5 is reserved to LED strips

What? I'm using only S6 and it doesn't output PWM.