iNavFlight/inav #5399: Docker compilation firmware problem

dybiusz, 2020-02-10:
Dear Friends,

Since this is my very first post here , so first of all thank you for your job in creating and developing of my loved INav. I am appreciated for your time and effort by making Inav and all related firmware.

I come here because I want to use my F4omnibus V2 pro in fixed wing plane with flaps and double channel ailerons.  In order to do that I would need 5PWM for servo and 1PWM for motor. Because in Inav there is not possible to remap resources like in betaflight I landed at customized firmware.
I found the post with description what to change in target.c , but got stuck at compiling this thing.

I have docker , atom and GITHub already installed. Have the path like "C:\Users\DELL\Documents\GitHub\inav\src\main\target\OMNIBUSF4\target.c"
so create bat  file to compile :
------------------------------------------------
docker run --rm -it -v //c/Users/DELL/Documents/GitHub/inav/src/main/target/OMNIBUSF4 :/opt/inav inav/inav-build

cmd /k
------------------------------------------------

If I try to compile I get message:
docker : invalid reference format

Could somebody help me and provide proper .bat text ? I would be very grateful for any help and hints.

Marcin

ShikOfTheRa, 2020-02-11:
Building in WSL is easiest option for Windows 10 - I can post some detailed instructions if you would like to try that option

namirda, 2020-02-13:
try something like:
docker run --rm -it -v "c/Users/DELL/Documents/GitHub/inav/src:/home/src/" inav/inav-build make TARGET="OMNIBUSF4"

DzikuVx, 2020-02-19:
In Windows it is recommended to use WSL instead on Docker. 
But even so, you commands seems to be wrong, refer to https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Docker.md

closed 2020-02-19 by DzikuVx