iNavFlight/inav #5401: CC3D Revo inav 2.4.0 mix does not work

ricardovalmeida, 2020-02-11:
hi dear friends.
I'm using the revo cc3d card with firmware 2.3.0 Nov 24 2019, inav version 2.4.0. for a tricopter.
I'm trying to mix the S6 for picht gimbal, but when I enable it, I don't have a signal output and my tricopter doesn't mount.
Can you help me thanks

[image]

ThiloBaWue, 2020-02-16:
I can confirm this problem. i am trying to use 2 Revolution FC´s for my Hexacopter. With Librepilot or Betaflight i get a PWM Output fo rmy Motors, but with INAV (which I want to use) I can´t get a PWM Output. tried nearly everything, but couldn´t find the solution. Yesterday arrived my Kakute F7 board and it worked without problems, with stable PWM on the output´s. I think there must be an bug in the Hex File for Revolution board´s. Perhaps the Pin Mapping?

ricardovalmeida, 2020-02-17:
hi dear friend. I hope you can help us by fixing the revo cc3d card firmware.
I have a tricopter and is flying very well with the inav 2.4.0, I wish I could add the gimbal.I made my tricopter on the 3D printer, if you want I can share the construction files.

ThiloBaWue, 2020-02-19:
Hi ricardovalmeida! I try my best, but I am not an perfect programmer (only Arduino User with medium skills). Looked at the Revo Code and compared it to betaflight. Thought the differences would be little, but it looks mostly different. Maybe I can do a little bit investigation, but perhaps there is an better programmer out there? And I go skiing for a week, so it will last a while till I have enough time.

closed 2020-05-03 by stale[bot]