iNavFlight/inav #5403: What happens when a prop fails?

abusr, 2020-02-12:
I'm surprised not to find something relevant. Maybe I wasn't lucky searching. I was trying to find the behavior of a quad with inav when a prop/motor fails. Does inav cover this case?

giacomo892, 2020-02-12:
Drone will spin out of control and crash

closed 2020-02-12 by giacomo892

abusr, 2020-02-12:
Provided sufficient thrust from the remaining engines why isn't it possible for the quad to land, even in a spinning condition? This could help limiting the total damage a bit, or not?

giacomo892, 2020-02-12:
It would be possible using advanced algorithm to land when spinning.

rsilk1949, 2020-02-17:
Build a hex copter ;)

DzikuVx, 2020-02-17:
Quadcopters need 4 motors and propellers to be working to maintain stability. This is the "hardware limitation" that can not be fixed by any software. 4 actuators are required to balance roll pitch yaw and thrust