iNavFlight/inav #5405: Mag alignment with cli not possible

brainbubblersbest, 2020-02-12:
Since a few Days, multiple Problems occured in Inav Facebook Group.
Problem: Users who mounting Magnetometer with Compass on a 30degree Tilt Angle Backwards, are not able to get this working with set mag_align_tilt = 300 or -300.

Lot of them build a dozen quads with inav and tiltet mag/gps. And now got it only working with a flat mount magnetometer. (EG bn880 with cw270 flip and mag_align_pitch = -300

Or mean this:
// on-board mag alignment. Ignored if externally aligned via *DeciDegrees.

That cw270flip would be ignore and you need to set
Mag_align_pitch = -300
mag_align_roll = 1800
mag_align_yaw = 2700

https://github.com/shellixyz/inav/commit/1f8d73c638facfcf07d2ca959f06132fee69f517

Bednarov, 2020-02-12:
I also have a problem with magnetometer. I want to mount it on a quadcopter. Angled backwards so when I fly the GPS looks directly up. It allows the unit to see more satelites constantly.

When the GPS+Compass unit (BN-880) is flat, it works perfectly. When it's tilted backwards, after calibration the heading readings are messy.

I tried setting align_mag_pitch to -300, 300 (for 30* angle that I have), calibrating after every change. Also i tried replicating the "CW270Flip" setting, in case that it's ignored when using the pitch angle cli command. Nothing works. I tried current firmware and also INAV 2.2.0

I was calibrating on a 2m cable, so no interference during calibration, also no interference from the quad, because when mounted flat in the same spot mag works perfectly. The tilt is the problem.

brainbubblersbest, 2020-02-16:
Hello, just wanna say. I found this in the betaflight Code. That mean cw270 flip mean flip over pitch axis 180 value and rotate over Yaw axis 270 value=
set mag_align_pitch = 1800
Set Mag_align_yaw = 2700

Means for 30degree backwards tilted
Reduce pitch value on 300
Mag_align_pitch = 1500

 #define CUSTOM_ALIGN_CW0_DEG         ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0,   .yaw = 0    }})
#define CUSTOM_ALIGN_CW90_DEG        ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0,   .yaw = 90   }})
#define CUSTOM_ALIGN_CW180_DEG       ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0,   .yaw = 180  }})
#define CUSTOM_ALIGN_CW270_DEG       ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 0,   .yaw = 270  }})
#define CUSTOM_ALIGN_CW0_DEG_FLIP    ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 0    }})
#define CUSTOM_ALIGN_CW90_DEG_FLIP   ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 90   }})
#define CUSTOM_ALIGN_CW180_DEG_FLIP  ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 180  }})
#define CUSTOM_ALIGN_CW270_DEG_FLIP  ((sensorAlignment_t){ .values = { .roll = 0, .pitch = 180, .yaw = 270  }})

 void buildRotationMatrixFromAlignment(const sensorAlignment_t* alignment, fp_rotationMatrix_t* rm);
void buildAlignmentFromLegacyRotation(sensorAlignment_t

brainbubblersbest, 2020-02-16:
Maybe this short hint can be added to the inav Wiki?

closed 2020-02-17 by brainbubblersbest

alienwaredream, 2020-06-09:
Hello everyone!
I found something not right in the those CLI
in INAV2.5.0 should be ：
set align_mag_roll = 300
set align_mag_yaw = 900
set align_mag_pitch = 1800

brainbubblersbest, 2020-06-09:
And what value is wrong?

alienwaredream, 2020-06-09:
**align_mag_roll
mag_align_roll**

As a rookie, I am a little confused. @@！

brainbubblersbest, 2020-06-12:
Yes, you are right maybe. Thats taken from the betaflight Free mag
alignment issue. Where These cli variables are correct. I Look over it
tomorrow.

brainbubblersbest, 2020-06-12:
Im confused too because These cli variables at betaflight are correct. So it seems inav use the variables as you describe. I Look about that tomorrow.

brainbubblersbest, 2020-06-13:
Should be correct now
Bf use also mag_align "custom" which is not existent in inav. Some Times it's confusing if you use multiple inav and bf fcs 😅