iNavFlight/inav #5406: Max throttle defaults to 1850, GUI element to fix it missing in 2.4.0 configurator

ShawnzAZ, 2020-02-13:
It could be I just couldn't find it, but max throttle seems gone from the GUI.

You can of course fix it by setting max_throttle=2000 in the CLI.

This is just annoying since the default is 1850 vs 2000.  Easy enough to fix via CLI, but figured I'd mention it since max_throttle wasn't mentioned in the release notes.

Inav version is 2.4.0

# dump

# version
# INAV/MATEKF405 2.4.0 Feb 11 2020 / 22:43:00 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1494 -100
servo 4 1000 2000 1459 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0

[... 1151 lines cut ...]
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set manual_rc_expo = 0
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 1050
set battery_capacity_warning = 210
set battery_capacity_critical = 157
set battery_capacity_unit = MAH

# end the command batch
batch end

DzikuVx, 2020-02-13:
there is no reason to ever change max_throttle setting. For DSHOT it does not matter and for other ESC protocols it is advised to calibrate ESC to defaults value

closed 2020-02-13 by DzikuVx

ShawnzAZ, 2020-02-13:
Fair enough, but you are losing throttle resolution.  This seems like it would be a larger problem with multirotor.

Thanks for the update though!

DzikuVx, 2020-02-13:
Escs does not have that much resolution on output side as we would like to
believe .

trailx, 2023-10-27:
I ran into an issue related to this on a wing with a Matek F411-WTE, running INAV 6.1.  I had to swap ESCs (the FC board was also changed out, but CLI was copied over and I didn't adjust max_throttle).  The old ESC must have been calibrated for 1850, and the plane flew great.  The new ESC must have been calibrated to 2000 max, and the plane barely took off, and could barely maintain altitude at NAV cruise settings due to the scaling in the throttle.  At least - that's my only explanation so far.  Used to cruise well at 60% power, and now it required around 80% power to attain the same speeds and current draw.

Is 1850 the standard for throttle mapping with ESCs?  I would have assumed 2000 is the norm.