iNavFlight/inav #5408: [request] Add minimum flight data logging for flight history

b14ckyy, 2020-02-14:
## Current Behavior
The blackbox feature on SD card is fantastic to analyze the behavior and fine tune a lot of stuff. But even with the lowest logging rate of 3% (30Hz at a 1k loop time) the logs get pretty big on long fligts. 

## Desired Behavior
After all the tuning and feature configuration is done, I'd like to use the Blackbox only for simple flight recording and flight history, to archive with my videos and recheck where every flight and moment in the video exactly was. Therefore a very low logging rate would be fine as an option just for flight recordings and archives. 

## Suggested Solution
Add a separate Flight recording option that sets the logging rate to 1Hz or 2Hz logging rate only. This will shrink down logs massively. 

## Who does this impact? Who is this for?
This will solve the issue of big logs replayed in MWPTools (@stronnag) causing the PC to slow down until it is unusable after rendering 54.000 log points in a 30min flight at lowest logging rate. Right now it is impossible to replay 60min+ flights in MWPTools due to the massive amount of logged points.  It will also make it much easiert to show tracks in Google Earth's 3D view as it gets similar performance issues for long flights. 

## Additional context
None

stronnag, 2020-02-14:
You can already do this, see https://github.com/iNavFlight/inav/pull/4302

stronnag, 2020-02-14:
> This will solve the issue of big logs replayed in MWPTools (@stronnag) causing the PC to slow down until it is unusable after rendering 54.000 log points in a 30min flight at lowest logging rate. [...]

This can (should) be addressed in mwp. The mwp option:
```
 -M, --mod-points=N                  Modulo points to show in GPS trail
```
will help, but a more useful approach would be for mwp to fetch points from the BBL less frequently than the 0.1s interval currently used regardless of log size. An mwp issue on this would be appropriate (and would be addressed).

stronnag, 2020-02-14:
https://github.com/stronnag/mwptools/issues/71 should solve the mwp part (and #4302 should have already addressed the inav part).

b14ckyy, 2020-02-14:
Oh I see. In CLI much higher values are possible than in GUI. Did not check that. Thank you very much!

b14ckyy, 2020-02-14:
OK the change in MWP works great with lower pull rate. INAV has the option to set lover log rate via CLI. So This issue is better addressed in the configurator and not in INAV. Closing here.

closed 2020-02-14 by b14ckyy