iNavFlight/inav #5412: Pixel osd still flashes on inav 2.4.0 stable

fmartinezo, 2020-02-16:
Here you have a video with the pixel osd with the stable version 2.4.0 of inav. https://my.pcloud.com/publink/show?code=XZV0eskZ27tVf3dBsmBLjNhtkGRkFyLYX10X
As you can see it has improved a lot since version 2.3 but something still flashes. 

## Additional context
I have also sent Alberto GH to his telegram a copy of the video
---
- FC Board name and vendor: MATEK F405-WING
- INAV version string: 2.4.0 stable

fmartinezo, 2020-04-17:
Am I the only one that this happens to? Or does it have no solution?

Seba-VT, 2020-06-17:
damn, your OSD looks awesome compared to mine... 

[https://www.youtube.com/watch?v=zhsbP_MNjMg](url)

gbine, 2020-06-25:
@fmartinezo  I'm struggling to get my Matek Analog Pixel OSD working on my F405 wing.  I get video passthrough but no graphics of any sort.  When I upload font, it appears to work (although it seems to be too fast for a 15200 baud connection)  is there a trick to disabling the on board OSD?  Did you have to do anthing unusual to get it to work?
Thanks

fmartinezo, 2020-06-26:
No, only follow manual

closed 2020-09-12 by stale[bot]