iNavFlight/inav #5414: Nav mode changes not seen in Tmp1 telemetry sensor

rsilk1949, 2020-02-17:
I am bench testing my first iNav model and now verifying all the flight modes are working correctly. Everything look good when I check the Radio reception and mode pages where I see the expected pwm, mode highlighting etc.

On my radio I am checking the telemetry and notice that Tmp1 changes as expected when selecting Acro, Angle and Manual (5, 15, 45). But when I select RTH or PosHold the Tmp1 value does not change.
I have armed and do have a good GPS fix as verified by the values in Tmp2 (e.g. 3811)

 I am now thinking that unless the model is actually flying, even though I can see iNav is receiving the change to a Nav mode, it is not executing it because it is still at home or too close to home or too low altitude. Therefore, it would only update Tmp1 when it actually executes the mode change.

Is this the case?

DzikuVx, 2020-02-19:
The event has to be actually happening to update the Tmp1 values. Selecting it via RX is not enough

closed 2020-02-19 by DzikuVx

rsilk1949, 2020-02-19:
Tx Pavel for confirming.