iNavFlight/inav #5416: Profile switching via RC adjustments non functional

Arvycka, 2020-02-17:
## Current Behavior
Currently rate profile switching via RC adjustments is not implemented in INAV

## Steps to Reproduce
1. Setup Rate profile switching via a 3 pos. switch in Adjustments tab
2. Change switch positions - rate profile is not changed.

## Expected behavior
1. Rate profile is changed depending on the switch position on the radio transmitter.

## Suggested solution(s)
Fix rate profile switching via RC adjustments function.
OR
Remove this option from INAV configurator and INAV manuals.

## Additional context
Rate switching currently is working in Betaflight and was working in Cleanflight but has never been enabled in INAV, yet it's still present in the latest INAV configurator and described in the manuals.

- INAV version string: 1.6-2.4

DzikuVx, 2020-02-18:
I'm in favor of dropping this as part of Adjustments and rebuilding in a different way

Arvycka, 2020-02-18:
It looks like nobody apart me has ever event tried to use this function 🙂
Do you mean switching, not rebuilding, in a different way.

DzikuVx, 2020-02-18:
The mechanism to load new profile in flight is broken atm. And the way how adjustments work is also a nightmere. So I'm inclining towards doing this with logic conditions and global function. And of course fixing the profile load in flight

Arvycka, 2020-02-18:
Agreed. Logic conditions sound like the way to go. 
I tried to fix it myself, even though I'm not a programmer, but gave up. Reworking it completely is also not something I would be able to do.

Martijn0001, 2021-03-31:
I also tried to use rate profiles changes by switches. Looking at the CLI, rate_profile doesn't even exist anymore. If this is reworked, can you maybe consider also placing nav_mc_hover_thr in the profile? This will help in switching between a 4s and 6s pack.