iNavFlight/inav #5417: nav_rth_tail_first in waypoint RTH ignored

Jacky2k, 2020-02-17:
## Current Behavior
When setting an RTH waypoint the value of nav_rth_tail_first is ignored and when turned on the craft still flies nose first to me.

## Steps to Reproduce
1. Turn on nav_rth_tail_first
2. Create a waypoint mission with one waypoint a bit away from you and after that another RTH waypoint.
3. The craft will fly nose first towards you for the RTH waypoint.

## Expected behavior
Fly tail first if nav_rth_tail_first is set.

## Suggested solution(s)
I guess (have not looked at it yet in detail) the normal RTH mode and a waypoint RTH are handled by different parts of the code as an waypoint RTH is still a normal waypoint and not a mode on its own and other waypoints may follow. So the part for WP RTH is missing the check for nav_rth_tail_first and should suppress NAV_POS_UPDATE_BEARING to setDesiredPosition, if it is set.

---
INAV/MAMBAF405 2.4.0 Feb 15 2020 / 20:25:51 (6c490a0fb)
GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

hali9, 2020-02-18:
> normal RTH mode and a waypoint RTH are handled by different parts of the code

WP RTH and RTH - this is a separate code. 
WP RTH not implement all future of RTH.

Jacky2k, 2020-02-18:
OK, so there is no reason to not implement that?

closed 2020-05-02 by stale[bot]