iNavFlight/inav #5419: Akk omnibus f4 V2, just work 1 servo!

Chewpasta1, 2020-02-17:
I have just received an akk omnibus f4 v2, and I am trying to setup my zohd dart with inav. So far I have flashed INAV 2.4 onto the board and gotten the gps to work and the esc too, the reciever tab shows inputs. When I use the customairplane to do mixing, no matter what servo number I put it always moves the same servo, if i change the input to the other sevo it work too, but i can´t figure out how to set up the 2 servos. if someone can help me please!

[image]
[image]

sorry the mess 
[image]

ports
[image]

DzikuVx, 2020-02-19:
please attach a full dump of configuration

Chewpasta1, 2020-02-19:
> please attach a full dump of configuration

# version
# INAV/OMNIBUSF4PRO 2.4.0 Feb 11 2020 / 22:14:01 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 -100
servo 2 1000 2000 1500 100
servo 3 1154 2000 1100 -100
servo 4 1154 2000 1100 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0

# gf
gf 0 0 0 0 0 0 0
gf 1 0 0 0 0 0 0
gf 2 0 0 0 0 0 0
gf 3 0 0 0 0 0 0
gf 4 0 0 0 0 0 0
gf 5 0 0 0 0 0 0
gf 6 0 0 0 0 0 0
gf 7 0 0 0 0 0 0
[... 1142 lines cut ...]
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

thanks

tiriad, 2020-02-20:
Try to configure a motor on output 2 even if you do not use it, in the same way as S1. I had understood that all the outputs were consecutive, so with your current configuration you are using PWM1, PWM2 and PWM3

closed 2020-03-24 by DzikuVx