iNavFlight/inav #5432: Motor output S5 does not work on Matek F722SE

Base-0ne, 2020-02-20:
Hi,
I'm using an Matek F722SE Flight controller with Inav 2.4 for an Hexacopter build. (Hexa-X)
With this setup Motor 5 will not work. i switched motor 5 and 6 on the FC and then Motor 6 wouldn''t move. 
I tested this on 2 FC''s (Brand new MatekF722SE) both has the same problem.
When installing Betaflight on the FC all motors work.

Is there an fix for this problem?
Or do i have to experiment with older Inav versions?

I hoped to remap the motor 5 to S7 but thats not posible in Inav 
(i did not have to do this in Betaflight)

[Dump 19-02 Hexacopter F722SE.txt](https://github.com/iNavFlight/inav/files/4231706/Dump.19-02.Hexacopter.F722SE.txt)
[BTFL_cli_20200219_215025 (Orgineel F722SE).txt](https://github.com/iNavFlight/inav/files/4231715/BTFL_cli_20200219_215025.Orgineel.F722SE.txt)

JeanFPV, 2020-02-20:
http://www.mateksys.com/?p=5159#tab-id-7
INAV Target Tips
S5 doesn’t work for Hex/Octo motor due to normal DMA confliction if Dshot is selected
Use multishot for Hex/Octo copter
BF uses burst DMA, no confliction between S1 ~S8 Dshot

Jacky2k, 2020-02-21:
Don't know exactly, but does https://github.com/iNavFlight/inav/commit/60c13ef71bc07aeb7354127aeada8bf49c61d82f fix that?

DzikuVx, 2020-02-24:
For hexa in INAV you can not use DSHOT on this board. Use Multishot instead

closed 2020-02-24 by DzikuVx

Base-0ne, 2020-02-25:
Thank you very much! I will put it on Multishot.

Sebbb, 2020-12-27:
Can somebody explain me what the background is? I don't understand what "normal DMA confliction" means. Am I able to get S5 working with DSHOT if I compile the firmware myself and e.g. assign a different timer? Or to use a different pin than S5 to get 6 DSHOT outputs?