iNavFlight/inav #5433: Fixed wing servos freeze on 2.4.0

AlistairR75, 2020-02-21:
## Current Behavior

## Steps to Reproduce

1.Flight controller in Matek 722wing fitted to skywalker

2.When armed move sticks rapidly to activate servos. They move as expected for close to 10 sec then freeze at end points. We thought this was caused by to much draw on Matek BEC so half servos were powered from separate BEC. Problem continued for all servos driven from FC. With Output Tab open while servos frozen the GUI shows normal movement following stick commands but servos are still frozen. 
3.We copied "diff all" from CLI and reflashed with full erase to firmware 2.3 and reloaded "diff all".
Problem was gone. Servos continue to work as expected and flight was normal.
We copied "diff all" from CLI and reflashed with full erase back to firmware 2.4 and reloaded "diff
 all". Problem returned
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

statherosjohn, 2020-03-09:
Yes i think its a bug, i also noticed when you move all controls fast and together at the endpoints , the flight controller freezes . I flashed 2.3 version and its warking fine with the same diff loaded.

digitalentity, 2020-03-10:
Please provide a config dump so we can debug and narrow this down

statherosjohn, 2020-03-10:
# version
# INAV/MATEKF405SE 2.4.0 Jan 30 2020 / 22:49:38 (7507cb140)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 2 0 -100 0 -1
smix 1 3 14 -100 0 -1
smix 2 4 1 -100 0 -1
smix 3 5 2 100 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0

# gf
gf 0 0 -1 0 0 0 0
gf 1 0 -1 0 0 0 0
gf 2 0 -1 0 0 0 0
gf 3 0 -1 0 0 0 0
gf 4 0 -1 0 0 0 0
gf 5 0 -1 0 0 0 0
gf 6 0 -1 0 0 0 0
gf 7 0 -1 0 0 0 0

[... 1146 lines cut ...]
set yaw_rate = 21
set manual_rc_expo = 0
set manual_rc_yaw_expo = 0
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 7000
set battery_capacity_warning = 2800
set battery_capacity_critical = 1749
set battery_capacity_unit = MAH

# end the command batch
batch end

Write your command here

BarryOrph, 2020-03-10:
Hi, I have the same problem in 2.4.0 on MatekSys F722-WING.

However, I noticed that this problem occurs when the servo rudder is tilted around 60 on my radio in the tab Output-> Servos-> 5 (my rudder servo) is 1220-1280 with min 1000 max 2000. I will upload the video on YT.

# diff all

# version
# INAV / MATEKF722SE 2.4.0 Feb 11 2020 / 22:53:38 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mmix 0 1.000 0.000 0.000 0.000
mmix 1 1.000 0.000 0.000 0.000

# servo mix
smix 0 3 4 100 0 -1
smix 1 4 4 100 0 -1
smix 2 3 14 100 0 -1
smix 3 4 14 -100 0 -1
smix 4 5 6 100 0 -1
smix 5 2 5 100 0 -1

# servo

# logic

# gf

# feature
feature -TX_PROF_SEL
feature -TELEMETERS
feature -BLACKBOX
feature -AIRMODE
feature -OSD
feature MOTOR_STOP
feature PWM_OUTPUT_ENABLE

# beeper

# map

# series
serial 1 16 115200 115200 9600 115200
serial 2 2 115200 19200 0 115200
serial 5 64 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 4000
set gyro_sync = OFF
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6000
set acczero_x = 143
set acczero_y = -40
set acczero_z = -168
set accgain_x = 4074
set accgain_y = 4064
set accgain_z = 3984
set align_mag = CW180
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set throttle_idle = 12,000
set align_board_roll = -2
set align_board_pitch = -42
set align_board_yaw = 1800
set platform_type = AIRPLANE
set model_preview_type = 14
set small_angle = 180
set applied_defaults = 3
set inav_use_gps_velned = OFF
set i2c_speed = 100KHZ

# profiles
profiles 1

set mc_p_pitch = 20
set mc_d_pitch = 0
set mc_p_roll = 20
set mc_d_roll = 0
set mc_p_yaw = 45
set mc_i_yaw = 5
set mc_i_level = 5
set yaw_lpf_hz = 30
set nav_mc_pos_z_p = 40
set nav_mc_vel_z_p = 0
set nav_mc_vel_z_i = 0
set nav_mc_vel_z_d = 0
set nav_mc_pos_xy_p = 75
set nav_mc_vel_xy_p = 0
set nav_mc_vel_xy_i = 0
set nav_mc_vel_xy_d = 0
set nav_mc_vel_xy_ff = 0
set rc_yaw_expo = 70
set roll_rate = 6
set pitch_rate = 65
set yaw_rate = 9

# profiles
profiles 2

set rc_yaw_expo = 19
set roll_rate = 21

# profiles
profiles 3

set roll_rate = 6
set pitch_rate = 106

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profiles 3
battery_profile 1

# save configuration
save

BarryOrph, 2020-03-10:
https://youtu.be/jfXt8vUcU1s

digitalentity, 2020-03-11:
@BarryOrph, does the FC totally freeze, or you can still connect to it from the Configurator?

I think I have F722-WING in my desk, will set it up and try reproducing.

digitalentity, 2020-03-11:
@BarryOrph, so in your video FC is disarmed, roll/pitch centered, throttle low and at certain yaw-left position FC locks up?

BarryOrph, 2020-03-11:
FC is calibrated, disarmed (I don't have ESC connected) and is tilted at a random angle, I've never flown it. The throttle on the video is low, but it doesn't matter, I thought it could also affect this behavior, it doesn't. This situation arises at different throttle positions when yaw is at a specific value, and the second stick is moved randomly.

As for hanging the whole FC, here the situation is quite strange because in 90% of cases I had a connection and I could see live values ​​in the Receiver and Sensors Tab after suspension, but sometimes the FC restarted itself ~ 5% of the situation (green led was flashing) Sometimes after suspending the servos at some point after 4-8 seconds they tried to do something but they immediately hung up again and I had to reset FC.

It looks like a memory leak or division by 0 in some situations. Although then this problem would not only apply to the wing controller. Hmm.

I did a downgrade to 2.3.0. If you have any scenario to test, go ahead.

BarryOrph, 2020-03-11:
...yaw-left is at a specific value... I have the impression that yaw-left is **above** a certain value.

lichtl, 2020-03-13:
Is it possible that it went into OSD menu? Because I remember servo outputs have been disabled in there.

AlistairR75, 2020-03-13:
In our case the plane was armed.  It happened 10 seconds into maiden flight

AlistairR75, 2020-03-13:
And it has been able to connect to GUI. outputs tab so servos should be moving.

BarryOrph, 2020-03-13:
@lichtl  I was also thinking about OSD. I turned it off for testing in the configuration tab, it did nothing.

digitalentity, 2020-03-16:
I can't reproduce your scenario, @BarryOrph. When board is ARMed I can't repro the freeze regardless of stick position. On disarmed board the FC enters CMS menu at a certain stick combination (Pitch up + Throttle Mid + Yaw Left). Interestingly, this is also active when OSD is disabled, therefore you can't exit it because you can't see the menu.

@AlistairR75 can you share your configuration as well? Is your board armed when you reproduce?

digitalentity, 2020-03-16:
Also see #5509

DzikuVx, 2020-04-24:
I'm also not able to reproduce. Closing. If any need data appears, please reopen

closed 2020-04-24 by DzikuVx

b14ckyy, 2020-05-10:
A friend of mine actually discovered the exact same issue on it's talon GT with a F405SE. 
Adding video link: 
https://youtu.be/P7Uv4yzYCWU

I'll ask him for a dump too. INAV is 2.4.0.

He has added a 3300nF Capacitor to the 5V servo rail and the issue was fixed by that. 
maybe some interferences from some servos causing CPU hickups on 2.4?

CaptKrasno, 2020-08-27:
I had this same issue and was able to resolve it.  https://github.com/iNavFlight/inav/issues/6071

The CMS (on screen configuration menu) is activated by setting the throttle to mid, yaw left, pitch up. This was getting randomly activated when rapidly moving the sticks around.