iNavFlight/inav #5440: ESC beeps for around 2 mins after power on

MartinHugh, 2020-02-23:
Running iNav 2.4 on a Matek F411-Wing in a Volantrex Ranger G2

I flew this aircraft for a while without a FC fitted and the ESC/motor was fine when driven directly from the RX.

 have now put the FC in and for the first 2 mins or so the ESC beeps.  Once it stops, everything is fine, and it the ESC/motor work as expected.

However, there is obviously something which is not quite right.

Can anyone suggest what I might be missing?

Thanks

MartinHugh, 2020-02-24:
Redoing the ESC calibration twice more, seems to have sorted, this.  I had already carried this out as part of the original configuration.  When I redid it a second time, it seemed to cure it.  But then later when I re applied power the problem had returned.  I then did this a third time and it seems better.  It continues to beep for around 5 seconds and stops at the same time that the "hardware health" flag go green.

I will see how it fairs when I fly it.   If the problem returns I will try another ESC, but I would still be grateful if anyone has any insight on what may be going on under the bonnet to produce these symptoms.

DzikuVx, 2020-02-27:
looks like ESC goes into calibration routine at startup.

closed 2020-03-23 by DzikuVx