iNavFlight/inav #5442: RTH Backtracking WP and/or LOG

stuwe911, 2020-02-24:
## Current Behavior
In RTH the direct line to Home might not always be the best. Terrain in between or areas of avoidance may force you to fly around or as of today above it. Sometimes adding an extreme altitude gain to get safely over that mountain and wasting precious energy.

## Desired Behavior

- New mode **RTH Backtracking WP**
- New mode **RTH Backtracking LOG**

## Suggested Solution
**RTH Backtracking WP**: If RTH is initiated turn around to the last WP passed and follow the mission in reversed order until reaching Home with the correct preset altitude for each WP.

**RTH Backtracking LOG**: If RTH is initiated turn around and use the flights log files to exactly fly the path in reversed order and altitude. No WP setup is required in this mode. Just fly the same way back as you came. This is intended for the more freestyle mission in difficult terrain.

## Who does this impact? Who is this for?
For all long range pilots that fly in difficult terrain or around airspaces, obstacles, aerea of avoidance etc. where the direct and shortest track to home is not the ideal situation especially to avoid extreme altitude settings. The altitude setting is no longer a concern. The way home is "safe" because it's the same how I got there.

## Additional context
Either of the modes would do to solve today's issues with RTH in the mountains.

hali9, 2020-02-24:
It looks similar to https://github.com/iNavFlight/inav/issues/962

stuwe911, 2020-02-26:
Indeed it does. thanks for the hint. i'm very happy to see that it's being worked on. I'll try to close or delete my request.

closed 2020-02-26 by stuwe911