iNavFlight/inav #5444: Enabling BLHeli32 Configuration

MaxLansing, 2020-02-24:
## Current Behavior
Unable to connect to the BLHeli32 ESCs under iNAV. 
Works fine when Betaflight is installed on FC.
(Am i missing something/doing something wrong? If so, can someone tell how to solve it?)

## Desired Behavior
Same functionality as under BF.

## Suggested Solution
Since i can't tell where the problem lies, can't suggest a solution. Sorry.

## Who does this impact? Who is this for?
Everyone running BLHeli32 ESCs i guess.

## Additional context
Hardware used:
Holybro F7 V1.5 and Tekko 32 F3 4in1 40A ESC
iNAV 2.3.x and 2.4.x

Thanks and Cheers to Developers!

DzikuVx, 2020-02-26:
Just tested with Kakute F7 V1.5, Tekko32 4in1 35A, INAV 2.4 and BlHeli32 709
Connected to the ESC without any problems, everything works like expected

closed 2020-02-26 by DzikuVx

MaxLansing, 2020-02-26:
Thanks Pawel for confirming.
For every one else, might having a similar issue:
When connecting with iNAV on the Board, i had the message : initialization of flight controller denied. (i haven't had the issue in BetaFlight and Dronin)
Whilst using betaflight worked just fine on the same hardware.
I don't know why, must have been from my starting days with iNAV, my Motor outputs where at 1003 at the lowest setting. after digging around i found an old post from BF 3.x times, mentioning a similar Problem. 
Setting the min_command=1000 solved the connections issue with the BLHeli Suite. 

Making noob errors i guess.... so long m8s.