iNavFlight/inav #5447: Barometer's reading keep increasing but drone stay on the table.

chan2328, 2020-02-25:
Barometer's reading keep increasing but drone stay on the table.
I cannot use ALT hold for my drone by the Barometer ( BMP280) on Matek 722 STD, so I buy a new barometer(MS5611), but the situation is the main unchanged, when I turn on the ALT HOLD the Drone goes up 5 meter, and sometimes goes down and land on the ground.

# status
System Uptime: 4255 seconds
Current Time: 2020-02-25T15:59:38.428+00:00
Voltage: 0.31V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x20010000, Heap available: 1820
I2C Errors: 45, config size: 5350, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = ADC 3
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 3, cycle time: 1009, PID rate: 991, RX rate: 49, System rate: 9
Arming disabled flags: RX CLI

# dump

# version
# INAV/MATEKF722 2.2.1 Jul  3 2019 / 22:40:11 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
[... 1110 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 420
set vbat_max_cell_voltage = 410
set vbat_min_cell_voltage = 280
set vbat_warning_cell_voltage = 300
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

chan2328, 2020-02-25:
[image]
[image]
[image]
[image]

JebKerman91, 2020-03-24:
I'm having the same issue with my Kakute F7 flight controller, which has a built-in BMP280:
[image]

[image]

JebKerman91, 2020-03-24:
Additional info: it seems to stabilize somewhat after ~half an hour of being on. Could this be due to the FC board itself heating up?

chan2328, 2020-03-25:
> Additional info: it seems to stabilize somewhat (while still climbing) after ~half an hour of being on. Could this be due to the FC board itself heating up?
GPS & magnetometer flag are on. Did you choose right GPS & magnetometer model for them? 

Could this be due to the FC board itself heating up?
no. I also use external barometer such as MS5611. the result still the same as BMP280 (climbing)

digitalentity, 2020-03-25:
Is it reproducible with an apartment window opened?

chan2328, 2020-03-26:
> Is it reproducible with an apartment window opened?

I tested it inside my home which on 25th floor with window opened.

brainbubblersbest, 2020-06-06:
can you all try set i2c_speed = 200 and report back?

Beardyblair, 2020-06-07:
|I have tested with 200, 400 and 800 and can confirm that I have alt hold working well. I am using a Matek F722-PX. I note that you have not set hover throttle or adjusted deadband from Default. Irratic or climbing "hover" can be due to this and can be exagerated by the craft setup. eg over or underpowered craft.

chan2328, 2020-06-11:
> can you all try set i2c_speed = 200 and report back?

Tried, same result.

chan2328, 2020-06-11:
> |I have tested with 200, 400 and 800 and can confirm that I have alt hold working well. I am using a Matek F722-PX. I note that you have not set hover throttle or adjusted deadband from Default. [...]

I did not set hover throttle and deadband. just by default  how to set it well?

brainbubblersbest, 2020-06-11:
Can u test it on earth Level and compare if your baro Do the same? For Alt hold you need a working gps or a lidar sensor. The baro value should not affect your altimeter value. If your altitude readings in sensor tab is correct

closed 2020-08-31 by stale[bot]