iNavFlight/inav #5456: Somewhat strange gyro readings in black box entries

JeanFPV, 2020-02-27:
Look at gyro, motors (not raw)...
The first test log file and I'm at a loss.  I did everything for the first time, inserted a flash card and flew.  The flight was successful, but the record is very strange.  I did not insist anything other than a recording frequency of 1/2.  Inav 2.4, bbexplorer 3.3. Maybe need some special settings?
[https://we.tl/t-xhTRBTVPle](url)

DzikuVx, 2020-02-27:
according to the log its recorded with 1/1. Lower to 1/4 or even 1/8 and try again

closed 2020-02-27 by DzikuVx

JeanFPV, 2020-02-29:
Well, I set 1/8 and try again, and got the same results. Do you have any other suggestions?

JeanFPV, 2020-03-03:
@DzikuVx well, I set 1/8 and try again, and got the same results (Record 1/1 and same problems). Do you have any other suggestions?
By the way, for 20 min flying I have (2 Gb flash) no free space, where was file with free space more than 1.5Gb, and logs...
[LOG00008.TXT](https://github.com/iNavFlight/inav/files/4282051/LOG00008.TXT)

JeanFPV, 2020-03-03:
> according to the log its recorded with 1/1. Lower to 1/4 or even 1/8 and try again

Respectable @DzikuVx, I tryed again... I set BB rate 1/4, 1/8 and 1/32, in SLI this settings was ready, and in UI too. 

> # get blackbox
>blackbox_rate_num = 1
>Allowed range: 1 - 65535

>blackbox_rate_denom = 8
>Allowed range: 1 - 65535

>blackbox_device = SDCARD
>Allowed values: SERIAL, SPIFLASH, SDCARD

After I plug-in SD and try to make logs. I just armed drone and turned his in my arms, and what you think? I have logs with 1/1 rate, with same problem again. Raw motor probably have true value, but decoded 100%... I have a sets of min_command 1000, Idle-throttle 5%(1048), motor have a reaction from 1100 and higher, but in logs I have min value of 1150. 
In settings of BB have only 2 parametrs, it is a rate and SD-ready.
Rate is true, SD is ready. I can't understand, what I do is wrong :( 
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/4282539/LOG00001.TXT)
[LOG00002.TXT](https://github.com/iNavFlight/inav/files/4282540/LOG00002.TXT)