iNavFlight/inav #5466: MR Throw mode/Launch Nav

DBetic-Pilot, 2020-03-03:
## Desired Behavior
APM and a few other GPS/Compass combinations have supported a throw/drop launch mode for multirotors. Given the hardware requirements already in place for any of the MR NAV modes, a throw launch would be incredibly useful. If i'm not mistaken, the APM code requires sensor calibration on the ground, then waits for a 5m/s velocity and turns on motors when vertical velocity reaches zero (at the apex of the throw). 

Suggested follow-on mode switches would be either to a Nav-POS hold, or a NAV-WP.
Thanks!

brat002, 2020-03-03:
The feature is one of those that will never been used.

DBetic-Pilot, 2020-03-03:
Do you have any data to substantiate this? It is a widely used feature for the APM users, as it enables free-flight and launching without controller in hand. Has this been developed or explored previously to suggest that it would not be used?

brat002, 2020-03-05:
I can't imagine somebody will throw his expensive quadcopter. What the problem do you want to solve with this useless mode? You drone doesn't have landing leggs? 
Inav doesn't have any problems with takeoff manually.

DBetic-Pilot, 2020-03-06:
Thanks for your opinions, but there are plenty of use cases substantiating why this would be valuable, as indicated by requests in betaflight and implementation in AC 3.4. The utility of the mode, matched with the ability to use on much smaller vehicles running inav, make this combination even more useful than when using a pixhawk. Hopefully someone other than the Russian troll will consider this...

DBetic-Pilot, 2020-04-08:
Wondering if the APM code could just be adapted and pulled over to INAV for this functionality, rather than starting from scratch?

brat002, 2020-04-09:
You are free to implement it and whow us why the feature isn't useless This is open source!

closed 2020-06-22 by stale[bot]

coyotespike, 2025-01-25:
@brat002's poor attitude - 'show us why this isn't useless, or we won't merge your feature if you develop it' - has left iNAV behind ArduPilot, which does have this feature.

Multiple teams have implemented a feature launching a small quadcopter by dropping from a large quadcopter.

This issue should be reopened. @DBetic-Pilot I would be happy to collaborate on this.

DBetic-Pilot, 2025-01-25:
Thanks for the offer, but i ended up moving on several years ago, and just running ArduCopter on my quads and bypassing the angle and speed limits in the .yamls. End result was successful launches both by dropping from a host platform and by throwing the quad out a car window at 60mph and having it arm at the apex of the throw and proceed on mission. For what we were trying to accomplish, we learned to that AC features were stable enough for us, and (as you mentioned) the INAV community combative and insulting demeanor not worth trying to work with or invest in.

MrD-RC, 2025-01-25:
IMHO this is a dangerous feature. Throwing a quad means you are not in control of it. It's purely a "feature" for showing off. In  reality if you can throw it, you can take off from the ground. My vote would be to keep this out of INAV due to safety.

I don't really care if ArduCoptor has it. ArduPilot has many bugs and bad decisions. For example. Say you throw the quad and the EKF throws a tantrum. Which does happen. I saw it happen last week. Its not uncommon. That quad is now going to behave randomly and could quite easily hurt someone.

coyotespike, 2025-01-25:
Again, there are many use cases for dropping a quad, and in a safe manner. It is extraordinarily useful for working in teams or swarms of drones, to combine e.g. capabilities of fixed wing and quads. I have no interest in showing off, I am building tools not toys.

Many thanks @DBetic-Pilot, I will probably end up doing the same thing.