iNavFlight/inav #5467: Allow "LAND" WP type (along with "POSHOLD_TIME" & "JUMP" that are currently being implemented)

peteoz, 2020-03-03:
## Current Behavior
for WP Mission planning it's currently possible to select "Land" option only if the last WP is RTH WP  

## Desired Behavior
same as above but extend the "Land" option to the last WP of the Mission even when the last WP is not RTH WP

alternatively, less "invasive" option to introduce "Land" option to last WP only if the last WP is "POSHOLD_TIME" where "time" parameter is not "-1" (infinite)

## Suggested Solution
suggested solution is described above, basically asking to implement MSP/mw-nav compatible WP type "LAND" that should not affect any existing mission planner software 

## Who does this impact? Who is this for?
it's for all users that wish to extend mission planning to more complete set of MSP/mw-nav compatible WP types, in this specific case it would allow safer implementation of one way missions without radio signal (failsafe_mission = OFF)

with the existing implementation one way missions (even when set to "failsafe_mission = OFF") are already possible but upon reaching final WP (when last WP is not RTH WP) the aircraft will loiter indefinitely until the battery dies and the aircraft crashes (in the absence of radio signal or when the user is simply unable to reach the final WP in time to take over the control of the aircraft)

if "LAND" WP type was implemented it would be possible to plan such one way missions in a safer way, for example using "POSHOLD_TIME" as last WP giving the user additional time to reach the rendezvous point (final WP) and take over the control of the aircraft, and additionally in case of the user not being able to reach the rendezvous point the aircraft would attempt to land instead of running out of battery.

## Additional context
With my wings I'm already able to execute over 100km WP Missions and occasionally I send my wing on a mission where last WP is far away from the launch site that I then travel to by car and land the model manually. I only do this in the remote Australian outback, usually in the deserts or other harsh terrain where my wing travels to areas that are otherwise not reachable by car, while I travel in a different direction to a rendezvous point (my route is usually much shorter than the route the wing is planned to travel). However it is possible that I may get a puncture or other misfortune may cause delay and this implementation would just allow for safer outcome.

Jacky2k, 2020-03-03:
I would prefer to simply implement the Land waypoint type. Then you can set both the HOLD and LAND at the same position (or close to each other). Or did I get something wrong?

peteoz, 2020-03-03:
yes that's correct, I would also prefer it that way if it was just a new WP type on its own without any ties to other WP types, and then it's up to the user how he/she wants to use it among other WP types in planning the mission.

because for example at the moment option "land" is only available in combination with "WP RTH" type, so it's just that - an option or rather a sub-type of WPRTH type.

so yes, implementing it as a new WP type on its own would make more sense and allow for more flexibility.

DBetic-Pilot, 2020-03-06:
+1, this would be great for experimenting with delivery techniques as well, particularly as GPS precision continues to improve.  It would certainly allow longer strings of behaviors to be tied together.

Jacky2k, 2020-03-06:
Maybe I can start working on that next week. It should not be too hard to implement. I think you just have to copy a buch of states from the RTH+Land state machine, rename it and you are good to go ;)

stronnag, 2020-03-09:
Seems to be working ....

'Plane descends (circling, in 20knots Westerly) from 50m to 8m where the pilot takes control at LAND WP5, aborting the landing. 

In glorious technicolour https://youtu.be/3uY7CUo93e4

[image]

closed 2020-05-23 by stale[bot]