iNavFlight/inav #5470: Inav for Aerial Mapping

Cutbangrian, 2020-03-04:
Halo, I've been using inav for a quite long time. since 1.5 version and there is a lot of upgrades.

What I want to ask, 
Is there any chance to use INAV for aerial mapping? just like ardupilot?

I see a lot of community on my town are using APM to fly their autonomous plane and use it to mapping such a field, landmark, etc.

could inav do the same? I mean Ive been using inav waypoint and it works amazing.
but one problem appeared, Inav doest have a feature to setting a shutter speed in a camera.

in ardupilot, they use mission planner to configure it. like take a picture every 50 meters, then log its coordinate.

after flying, all we need to do is match the coordinate with the picture that has been taken. and with Arcgis software, we can make a layout of a field only with those picture.

could inav do that?

swissfreek, 2020-03-21:
This would be a pretty cool feature to add to the mission planner (to plan the grid like Mission Planner or DroneDeploy can). The second step would be to add a way to trigger an event (something in the global functions/logic variables perhaps?) so that you could hook a wire up to a camera to snap a picture on command.

stronnag, 2020-03-21:
There are a couple of extant (and long standing) options for area / grid planning:

http://www.iforce2d.net/surveyplanner/
mwp area planner https://github.com/stronnag/mwptools/wiki

swissfreek, 2020-03-21:
You are right, I had completely forgotten about iforce2d's site (man, he's been around forever). Yours I was not familiar with. I will look into it.

stronnag, 2020-03-21:
I "borrowed" iforce2d's algorithm, they both give the same result.

Cutbangrian, 2020-03-21:
> This would be a pretty cool feature to add to the mission planner (to plan the grid like Mission Planner or DroneDeploy can). The second step would be to add a way to trigger an event (something in [...]

I knew about that, but this is about aerial mapping.

My friend used ardupilot, as a main job.
To locate and take a picture of a field that google maps doesnt care so much.

So they need to 'mapping' that area using autonomous plane.

So, in ardupilot they have mission planner to act just like inav configurator.

But, in mission planner, they have something or feature that could setting the camera to take picture every 5 or 10 meters and log the coordinate.

After that, they only need to join and match the coordinate that has been logged with the picture that has been taken.

So the final result is a map in satellite display.

Actually I can learn ardupilot, but I was too comfortable with INaV.

I hope you get my point.
In inav, we dont have that feature, and if I did just like you said, take a wire and switch to trigger the camera to take a picture,

How to set the camera that automatically take a picture in every 10 meters or every 10 seconds 😃

Cutbangrian, 2020-03-21:
> There are a couple of extant (and long standing) options for area / grid planning:
> 
> http://www.iforce2d.net/surveyplanner/
> mwp area planner https://github.com/stronnag/mwptools/wiki

Never heard about this but I will looking into that.

Is that possible to have aerial mapping with inav?

swissfreek, 2020-03-21:
Right, that's what I was saying. You can currently plan for a grid flight, but you cannot trigger the camera to take a picture at certain points because there are not waypoint functions yet. But it looks like it is coming as there have been recent pull requests to start working on having different waypoint functions the way Ardupilot does. Another nice thing Mission Planner does is all the math to figure out exactly where all the pictures need to be taken based on resolution and all that. I haven't had a chance to look into stronnag's program but if it does that, that's another step in the right direction. If not, it needs to be added somewhere, because it would be very tedious to create all the individual waypoints for all the pictures (and likely impossible to calculate the points by hand).

stronnag, 2020-03-22:
Neither of the extant area / grid planning solutions do anything more than plan a WP mission. Anything else requires flight controller support.

Cutbangrian, 2020-03-22:
> Right, that's what I was saying. You can currently plan for a grid flight, but you cannot trigger the camera to take a picture at certain points because there are not waypoint functions yet. But it [...]

seriously? is there any pull request about more functions of waypoint? thanks god. I really cant wait to see it.
So I could use inav for aerial mapping sooner.
basically this feature will work with simple principle right? just added more function on waypoint, like to trigger the camera and log the coordinate.

so far, inav is really good for me.

swissfreek, 2020-03-22:
> seriously? is there any pull request about more functions of waypoint? thanks god. I really cant wait to see it.
>
> So I could use inav for aerial mapping sooner.
>
> basically this feature will work [...]

I’m assuming that’s the eventual intent, but it looks like it’s fairly early in the process. I’m not sure there’s a ton of demand in the greater iNav user base for this sort of thing, so it might not be a priority.

a-425, 2020-04-22:
I use inav for mapping. But the output is only orthomosaic picture. We can set survey planner with inav mission planner for android. Then fly with camera facing down and record as video. From video file we can convert it into images (5 seconds oer frame as example). Then process it on agiSoft. Done.... 😁😁😁. But no gps data

Cutbangrian, 2020-04-22:
> I use inav for mapping. But the output is only orthomosaic picture. We can set survey planner with inav mission planner for android. Then fly with camera facing down and record as video. From video [...]

Seriously?
This is interesting. What type of plane?
How do you setup camera to triggered? 
Do you used CHDK script?

5 second interval I thougt were too long.
You should try 2 second interval.

Can I see the images?

To be straight, left me a contact so we can chat in private. Thanks mate!

a-425, 2020-04-22:
You can even make it 5 images per second. I press record before my quads take off. Here is my screenshot 
[image]

a-425, 2020-04-22:
I mean you can set as you want. 1 image per 2 second or 1 image per second or evem 3 images per second you are able to do that

a-425, 2020-04-22:
> Seriously?
> This is interesting. What type of plane?
> How do you setup camera to triggered?
> Do you used CHDK script?
>
> 5 second interval I thougt were too long.
> You should try 2 second interval.
>
> Can I [...]

Mine is 5 inch multicopter

a-425, 2020-04-22:
> Seriously?
> This is interesting. What type of plane?
> How do you setup camera to triggered?
> Do you used CHDK script?
>
> 5 second interval I thougt were too long.
> You should try 2 second interval.
>
> Can I [...]

ajiz.425@gmail.com

DzikuVx, 2020-04-22:
In INAV 2.5 it will be possible to program a servo output to for trigger with defined frequency. Trigger based on traveled distance is not confirmed

Cutbangrian, 2020-04-22:
> In INAV 2.5 it will be possible to program a servo output to for trigger with defined frequency. Trigger based on traveled distance is not confirmed

Really? I cant wait to see it!
Step by step we will race ardupilot.
There are Inav rover and boat.

Now triggering camera via Inav 2.5.

No problem with travelled with distance, step by step.

I hope in inav 2.5 we can hook up a camera and trigger it automatically, so I can use inav in a professional job of surveying and could contribute on the patreon. Cheers!

closed 2020-07-05 by stale[bot]

myrcwing, 2021-07-19:
Are they any updates whether camera triggering can be done in inav3.0?

it would be a great feature and open up INav to so much more.

swissfreek, 2021-07-19:
You should be able to make it work with global variables and logic conditions. For sure you can do it with a timer, and I'm pretty sure you can do it by distance as well now. If you look up Pawel Spychalski's YouTube channel, or Mr. D - Falling With Style, I think they both have made videos on global variables and logic conditions (you'll have to go back in time a year or maybe even two to find them, or search). It won't be nearly as user-friendly as the system that ArduPilot Mission Planner has, but it is feasible with some work.

The best place to ask for help or find information is definitely the iNav Fixed Wing Group on Facebook (it will come up in a search). Both of those gentlemen are active in there, as well as many others, and your question will get answered. Here you are limited to hearing from devs and the few of us nerds that follow conversations on GitHub.

myrcwing, 2021-07-19:
> You should be able to make it work with global variables and logic conditions. For sure you can do it with a timer, and I'm pretty sure you can do it by distance as well now. If you look up Pawel [...]

Thank you for the response with some info. I’ll check them out.