iNavFlight/inav #5471: Tramp protocol, new VTX handling, "Low power until armed" power level handling

DzikuVx, 2020-03-04:
## Steps to Reproduce
1. Matek HV-VTX
2. Tramp protocol configured
3. "Low power until armed" to anything besided "Disarmed"
4. Arm

## Current behavior
* VTX selects correct power level
* Power level LED keeps blinking
* VTX reboots when saving new configuration every 15 seconds forever

## Expected behavior
* VTX selects power level
* VTX reboots only once

digitalentity, 2020-03-09:
Weird. I don't have a HV-VTX, but I checked on another Tramp-compatible VTX and power level is sent to the VTX just once, as expected, no repeating every 15 seconds.

 #5461 changed the VTX polling schedule a bit, this could have triggered some edge case. I guess it's time to get that VTX.

digitalentity, 2020-03-09:
Actually, Tramp protocol itself is a mess. I should consider refactoring it.

digitalentity, 2020-04-07:
@DzikuVx can you please test https://github.com/iNavFlight/inav/pull/5575?

DzikuVx, 2020-04-13:
seems to be fixed by #5575

PR #5487 (2020-03-26): [VTX] Remove unused check; Fix Pit mode CMS
Should fix #5469
Removing extra check might fix #5471 as well.
closed 2020-04-13 by DzikuVx