iNavFlight/inav #5472: BLHeli Passthrough is Non-Op

kelleysislander, 2020-03-05:
## Current Behavior

Using the newly added Holybro Kakute f7 HDV target connected in the Kakute F7 HDV / Tekko32 F3 HDV ESC stack, I and was able to connect to the ESC's in BLHeli32 configurator via Betaflight on BLHeli Passthrough with BF 4.1.2 on the board... BUT, I was not able do the same with iNav 2.4.0 on the board.  I received the "Found no valid ESC configuration"
## Steps to Reproduce

1.  Flash iNav 2.4.0 and connect to the ESC and connect to the USB of the computer, get a port with USB Serial Device shown in Device Manager on Win 10
2.  Open BLHeliSuite32_32709 or BLHeliSuite32_32710 configurator and attempt to connect in the usual fashion.  You will not connect and will receive the "Found no valid ESC configuration" 
3.
4.

## Expected behavior

As a user I should be able to connect the ESC's via BLHeli Passthrough on iNav 2.4.0
## Suggested solution(s)

Restore the functionality
## Additional context

---

- FC Board name and vendor:
- INAV version string:

DzikuVx, 2020-03-06:
@kelleysislander please provide INAV diff output!

kelleysislander, 2020-03-06:
Ok, sure...

I’ll do it later today after I get back home from work. 

Cheers,

Bill

kelleysislander, 2020-03-07:
Here's the diff output:

Entering CLI Mode, type 'exit' to return, or 'help'

# diff all

# version
# INAV/KAKUTEF7HDV 2.4.0 Feb 11 2020 / 21:45:10 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

# servo mix

# servo

# logic

# gf

# feature
feature -AIRMODE
feature DYNAMIC_FILTERS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 0 2097152 115200 115200 0 115200
serial 1 2 115200 115200 0 115200
serial 3 64 115200 115200 0 115200
serial 5 0 115200 115200 0 115200
serial 6 262144 115200 115200 0 115200

# led
led 0 5,15::AB:0
led 1 6,15::A:0
led 2 7,15::A:0
led 3 8,15::A:0
led 4 9,15::AB:0

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 11 V
osd_layout 0 1 19 12 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 13 12 H
osd_layout 0 9 24 10 V
osd_layout 0 11 8 12 V
osd_layout 0 12 13 12 V
osd_layout 0 15 17 11 V
osd_layout 0 24 11 11 V
osd_layout 0 28 1 12 V
osd_layout 0 30 2 6 V
osd_layout 0 32 25 12 V
osd_layout 0 106 0 1 V

# master
set looptime = 250
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_hz = 175
set acc_hardware = MPU6000
set acczero_x = 56
set acczero_y = -19
set acczero_z = -56
set accgain_x = 4061
set accgain_y = 4086
set accgain_z = 4034
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 7
set serialrx_provider = FPORT
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set throttle_idle =  6.000
set failsafe_procedure = RTH
set model_preview_type = 3
set applied_defaults = 2
set rpm_gyro_filter_enabled = ON
set rpm_gyro_min_hz = 75
set mc_airmode_type = THROTTLE_THRESHOLD
set osd_units = IMPERIAL
set osd_rssi_alarm = 40
set name = Mr Croc

# profile
profile 1

set mc_p_pitch = 45
set mc_i_pitch = 35
set mc_d_pitch = 24
set mc_i_roll = 35
set mc_d_roll = 24
set dterm_lpf_hz = 75
set use_dterm_fir_filter = OFF
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set rc_expo = 50
set rc_yaw_expo = 50
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 50
set manual_rc_expo = 50

# profile
profile 2

set mc_p_pitch = 30
set mc_i_pitch = 50
set mc_d_pitch = 25
set mc_p_roll = 30
set mc_i_roll = 50
set mc_d_roll = 25
set mc_p_yaw = 60
set mc_i_yaw = 50
set dterm_lpf_hz = 80
set rc_expo = 50
set rc_yaw_expo = 50
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 50
set manual_rc_expo = 50

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 2
battery_profile 1

# save configuration
save

#

jadirc, 2020-03-07:
I had the same problem. But then I noticed that it always does not work when I connected to the INav Configurator before. If I connect the FC and then start BLHeli32 configurator directly, it works for me.

kelleysislander, 2020-03-09:
Thank you... tried that anew since you suggested... still unable to connect ;)  Did that work for you on the exact same setup as described above and pasted below?

 Kakute F7 HDV / Tekko32 F3 HDV ESC stack

jadirc, 2020-03-09:
I noticed this on a Matek F411 Wing FC with a Racestar 40A BLHeli32. Just thought this was a general behaviour.

DzikuVx, 2020-04-14:
Just checked with Kakute F7 (this is the same hardware minus OSD) and passthrough is working.
This is probably a timing problem: I2C processing for BMP280. I2C on 400khz takes too much time and messes passthrough timings. 
Your options:
* increace i2c_speed to 800khz
* or lower looptime to 500

kelleysislander, 2020-04-15:
Thank you for that solution. It potentially fixes myself and the two others that commented on the #5472 issue. 

How will all the other users know to look here to find the answer?  

Can you possibly build this fix into the FW for this problematic board?

Thanks,

Bill

tgreer, 2020-05-21:
@DzikuVx I have the F722-PX and have the same issue. I've set looptime to 500 and still no dice.

giacomo892, 2020-05-22:
INAV 2.5 on F722-PX
All working fine
[image]

kelleysislander, 2020-05-22:
The issue was raised on 2.4. 

Glad to see it works once more on 2.5.

digitalentity, 2020-05-22:
Reopening. 

In TG chat we figured out that BLHeliSuite32 1.0.0.2 is requesting motor config using `MSP_MOTOR_CONFIG` which is unavailable in INAV. BLHeliSuite32 1.0.0.1 was not doing this and thus is working fine.

@sskaug maybe you can chime in on this?

4712, 2020-05-22:
@digitalentity  BLHeliSuite32 xl/m 1.0.0.2 and 32.7.1.2 is restructured in some parts.
Need to dig into it ...

4712, 2020-05-23:
@digitalentity Fix will follow shortly in BLHeliSuite32 xl/m 1.0.0.3 and 32.7.1.3.

digitalentity, 2020-05-24:
Awesome, thank you so much @4712!

closed 2020-05-24 by digitalentity