iNavFlight/inav #5476: second I2C pins (SDA SCL) for MAMBA F405 fc

purpletreee, 2020-03-06:
Hello guys,
I wanted to add GPS functionality to my drone, so I ordered a betian bn880 gps with compass. Problem is, that for nav pos hold / alt hold you need a barometer which I realized later and by surprise the Mamba f405 does not have a barometer built in. But I acctually have an external barometer on hand, but there is only one set of SDA and SCL pins on the fc, and those are used by the compass. So my question is, if it is possible to use GPS altitude for althold, if not is there a way to add a second set of I2C pins to the board by replacing something like buzzer pin or any else, there are acctually a few spare pins on the MAMBA f405.

L4ky, 2020-03-06:
You can connect multiple devices to the same I2C pins, just solder them in parallel.

purpletreee, 2020-03-06:
right, I forgot. Well tried that now but I cant get the external bmp280 working, not even without the compass. Do I have to flash another target for an external barometer?

purpletreee, 2020-03-17:
Still not working any suggestions? I2C address 0x76

Jacky2k, 2020-03-20:
But the compass is working? So the bus is working in general (no short for example)?

purpletreee, 2020-03-20:
The compass is working so I can verify the port is working and the barometer aswell (checked it with an arduino). And I also ran an I2C scanner and found out the address is 0x76. Is yours the same? What is the default in inav?

Jacky2k, 2020-03-21:
Wow that is strange. I guess you also specified in the configurator that you are using bmp280?

purpletreee, 2020-03-21:
Yes I did that but I tried it on auto aswell

purpletreee, 2020-03-21:
I think it has something to do with the I2C address because I red somewhere that some bmp280 have 0x77 as address

purpletreee, 2020-03-21:
Mine is this one 

> https://www.amazon.de/Anpro-GY-BME280-barometrischer-Luftfeuchtigkeit-Temperatur/dp/B07LBKWSL1/ref=sr_1_6?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=bmp280+5v&qid=1584117670&sr=8-6&swrs=7DE6329639FA8744961B6A312D1FD605

>

Jacky2k, 2020-03-21:
> I think it has something to do with the I2C address because I red somewhere that some bmp280 have 0x77 as address

But you were able to address it using the arduino at 0x76?

purpletreee, 2020-03-21:
> 
> 
> > I think it has something to do with the I2C address because I red somewhere that some bmp280 have 0x77 as address
> 
> But you were able to address it using the arduino at 0x76?

correct

Jacky2k, 2020-03-21:
Mine is not exactly the same.. maybe my is the BMP180 - which is compatible.
However that makes complete sense. I looked a bit at the datasheet. Yes, the BMP280 can have either 0x76 or 0x77, depending on the pin SDO. If it is connected to GND, it is 0x76, 0x77 if connected to supply.
In file https://github.com/iNavFlight/inav/blob/master/src/main/target/common_hardware.c (line 96) the default address is defined as 0x76, so it should work in your case, as it does for me.
There could be a couple of faults:
* Wrong wiring. Double check that. Also make sure you have supply voltage at the sensor. It does have an LDO regulator on it, so you should connect 5V to it, but the SCL/SDA lines are 3.3V.
* Check that you have ~3.3V at SCL and SDA lines. There should be a pull-up on the bus (I am quite sure the MAMBA has them).
* You fried the chip with the Arduino (it cannot handle 5V levels, although that should not happen with I2C, when configured correctly - it is a open collector bus).
* There is something else wrong with the chip, so that it can be addressed but other stuff does not work. I think that is quite unlikely.
* Plug a logic analyzer to the bus and check what is going on.
* Make sure you selected that baro (set baro_hardware = BMP280).
* You could try to decrease the I2C speed (set i2c_speed = 100KHZ) in case you have very long wires.

MaxMihawk, 2020-04-01:
I have this combination working. It works out of the box. Check your wiring or change the BMP280.

quetiano, 2020-04-05:
> 
> 
> I have this combination working. It works out of the box. Check your wiring or change the BMP280.

Thats great! Did you use any resistors on the scl and sda when connecting the gps, compass and baro onto the mamba f405 ?

MaxMihawk, 2020-04-05:
No resistors or something else. I´ve just wired the Mamba to my BN-880Q (6 wires) and then from the BN-880Q to the BMP280 (4 wires).

Galgo13, 2020-04-05:
Hi guys.

I have a similar problem. I'm using an Flight colntroller F7 with an internal Barometer BMP280 and an external Magnetometer HMC5883. When I connect the mag. just physically to the scl/sda pins the baro isn't recocnized, the mag is recocnized. If I unsolder the mag. the baro is recocnized without any problem. 
But I haven't found a solution yet, even afer investing a lot of time. (GPS is working fine btw.)
Probaply the gps+mag. I bought is just crap...

@purpletreee: if you unsolder the mag. is your baro still not working?

What I'm corrently using:
FC: Mateksys FC F722-SE; Dual Gyro/Acc, OSD, Baro, SD Slot, PDB, Curr 184A, 5V2A
Baro: BMP280 on FC internal
GPS: APM2.6 2.8 GPS Module High Precision GPS With Electronic Compass PIXHAWK
•	GPS: UBLOX
•	Magnetometer: HMC5883

Jacky2k, 2020-04-05:
Well, if my tipps do not work, then I don't know anything else. You will have to use a logic analyzer and look at the code to find out whats wrong.

quetiano, 2020-04-05:
Thats going to be hard lol. I tried using three types of resistors, 3k, 4.5k and 10k, all did not make the compass work in inav, but in betaflight, just a normal connect like MaxMihawk,but, I remapped my tx 3 and rx 3 pins as scl 2 and sda 2, configured a few settings in cli in betaflight, and everything worked, but inav, only the gps and baro works, the built in compass with the gps never works. I tired those resistors in inav but none worked as well, seems inav just isnt letting my compass work, so I'm going to use the baro and gps. When this corona virus lock down is over here in SA, I am going to purchase another flight controller with a built in baro, so the gps with compass should work perfectly.

MaxMihawk, 2020-04-05:
I like the Mamba a lot. It is cheap, it is good quality and you have enought UARTS for everything you need for INAV.  My settings are:
UART1 - Serial RX
UART2 - TBS SmartAudio
UART6 - GPS
Softserial1 - Smartport (I´m using PPM for this)

Systeem:
Gyro & Loop Time = 2kHz

Sensors:
MPU6000, **QMC5883**, BMP280
Mag Alignment: **CW270° flip**
GPS = ON, UBLOX, Autodetect

You need a high distance (>10 cm) between the compass and your other gear.

You need to calibrate your compass befor you can use it. In the "Calibration" Tab on the right side "Compass Calibration". If you have done this, you have some numbers in the X, Y, Z boxes. You have to check, if you´ve done it right.

Some things are only working if you have a battery on your copter and a gps fix. So if your sensorhealth is red, this means nothing. Maybe you just don´t have a battery connected.

I have an maybe "exotic" gps (BN-880Q). If you have the common BN-880 you just go with the correct sensor name. My BMP280 are this Version: https://www.ebay.de/itm/BMP280-Luftdruck-Modul-Temperatur-I2C-Sensor-Barometer-Arduino-Raspberry-Pi-ESP/333356600262?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

quetiano, 2020-04-05:
I have... mag = HMC5883
             baro = MS5611
I did always have the battery connected whenever I tried to make the sensors work, they worked perfectly in betaflight, only thing is I had to remap in betaflight, in inav, theres no remapping that I know off. I have also tried all the names of the compasses available in inav, but none worked. The compass sign always remained red no matter what I tried, hence I gave up because to me, there was nothing left to try, but I managed to fix in betaflight. So I'm aiming for a new FC with built in baro.
Any other ideas to try ?

Jacky2k, 2020-04-06:
You can reduce the loop time somewhere or maybe increase the I2C speed. Are you sure the compass is not detected and not just not calibrated? I think it will show up red in the overview if it is not calibrated properly.

quetiano, 2020-04-06:
Its lit in red and the calibration button was greyed out. 
I've already tried setting i2c speeds, did not work.
I will try the loop times and earlier versions of inav because I've experienced earlier versions working well and the later versions dont detect certain sensors, perhaps an earlier version might work good.

Jacky2k, 2020-04-06:
Well, as I said, this is hard to remote debug when you cannot even do a record using a logic analyzer. Think about getting one (you can pic up Saleae 8 clones for less than 10$ on ali/ebay).

quetiano, 2020-04-06:
Thanks Jacky2k, the corona virus has us on lockdown so I cant even order anything for the next month lol. Even that new flight controller has to wait unfortunately. Thanks so much for your help so far.
I did not even know that there is something called a logic analyzer lol, just googling it now to learn more.

Jacky2k, 2020-04-06:
Hehe, you are welcome. For such low frequencies you can also build your own using an Arduino or a similar board. There is probably a lot of code for that purpose out there. You could probably also write some code for the F4 on the board to let it debug itself. You could even use the parallel port of an old computer ;)

quetiano, 2020-04-06:
Woooow!!!! What was that !!!! I had no clue what I just watched about logic analyzer and how to use it. Thanks so much Jacky2k, I'm happy if nothing else works with getting the compass to work, I'll just purchase a new flight controller with built in baro lol, after this lock down that is.

quetiano, 2020-04-06:
Do any of you have experience with other problems relating to quads ?

Jacky2k, 2020-04-06:
Sure, but this is actually not a chat here. Create a new report for every problem you have. Also you can checkout the Telegram group with almost 1000 people in it, there will probably be someone to help you.

quetiano, 2020-04-06:
Thanks Jacky2k.
I just tested all the sensors once again now, the baro is lit in red when both compass and baro is connected to scl and sda, I'm trying different configs with the i2c speed and changing the gyro and controller loop time, none seem to work unfortunately. Still trying...

Jacky2k, 2020-04-06:
There you go. There is currently a development version on it, so there might be a few settings that your firmware is not aware of - nothing to worry about.
[INAV_cli_diff.txt](https://github.com/iNavFlight/inav/files/4439080/INAV_cli_diff.txt)
[INAV_cli_dump.txt](https://github.com/iNavFlight/inav/files/4439082/INAV_cli_dump.txt)

quetiano, 2020-04-06:
Awwww thank you Jacky2k
I was wanting to compare and see the difference. 
I will keep you and anybody else following this in the loop, going to try to flash an earlier version now

quetiano, 2020-04-06:
Jacky2k, I have two questions,
1. Can altitude be adjusted for level hover without the use of a baro ?
2. Can Nav waypoints work without a baro ?

Jacky2k, 2020-04-06:
Well, theoretically you could use the GPS for that. But depending of the quality of the module, that might jump around a couple of hundred meters. And I cannot recall that there is any switch in iNav to support that. Without altitude there are no waypoints.
One thing that just came to my mind: Where did you connected the power supply of the sensors? Did you made sure that the voltages are correct? Usually (but certainly not always!) you have to connect those things to one of the 5V (and GND of course). Keep in mind that the pins labeled "Vcc" on the Mamba are actually battery voltage!

quetiano, 2020-04-06:
1. The mamba has only 5v soldering pads, I did not see any lower voltage than 5v on the flight controller. I'm using the stack mamba f405, which came with the 4 in 1 esc. I connected the baro to the tx/rx3 5v and ground soldering pads, and I connected the gps with built in compass onto the rx/tx6 5v and ground soldering pads

2. So, without a baro, I cant use NAV waypoints ?

Jacky2k, 2020-04-06:
1. What? The MS5611 has only I2C and SPI, you cannot connect that to a UART (aka serial, aka RX/TX). There are extra SCL/SDA pads for that. Also you hopefully know, I2C (aka SCL/SDA) is a bus, you can connect multiple devices in parallel.
2. I don't think so.

quetiano, 2020-04-06:
hahaha...Yes I understand, what I meant was I did not use the actual RX/TX pads, the rx/tx pads also have 5v and ground pins next to the rx and tx. I used those 5v and ground pins for baro and gps

Jacky2k, 2020-04-06:
Ahh yeah that is ok :)

MaxMihawk, 2020-04-06:
Make a photo of your wiring or mark it in the wiring sheme and post it here.

quetiano, 2020-04-07:
Good day guys. I've read up that the FC firmware can be extracted from the fc and modded to do pin remapping since inav does not support pin remapping. If I can mod the firmware or if one of you can help me on that, I should definitely be able to solve my inav baro problem, all i have to do is remap uart 3 (RX3 & TX3) to i2c_2 ports.

Jacky2k, 2020-04-07:
I don't know why that would help. I2C_1 is as good as I2C_2. Extracting the firmware and "modding" it is certainly possible, but a huge pain in the butt. Just clone the git repo and build the firmware yourself. At least in Linux that works perfectly with just a few commands. Don't know about Windows. Then you can change the mapping somewhere - can't help any further with that, I have never done that, because it is usually not required.
And you will have to do that procedure every time you want to use a new firmware release.

quetiano, 2020-04-07:
Greetings Jacky2k.
I've found what I was looking for! Ok, these guys were actually talking about the inav firmware, in its raw files, each file is written in C code I think, you can check for yourself, download the latest inav source file and you will see what I'm talking about.
I've found the file, fc_init.c ...this contains the i2c port configuration to use the uart 3 as i2c2 ports.

Now in your reply above, you mentioned that you dont see why that would help and i2c_1 is as good as I2C_2, ok, in betaflight, thats how I got both the compass and baro to work, the compass used i2c_1 and I remapped the uart 3 to i2c_2, and connected the baro to i2c_2(uart3), and both sensors worked well. My problem is that no matter what I tried in inav now, nothing got both sensors to work together when both were connected to i2c_1 only, so I have to try by splitting the baro i2c and giving the baro its own i2c port which is i2c_2, so no conflicts with the compass using i2c_1. It worked in betaflight and I have nothing left to try other than this, I'm all out already and if this doesnt work, I'm done trying, I'll wait for a new FC when the lockdown is over.

digitalentity, 2020-04-07:
I'm unable to reproduce on INAV 2.4, BMP280, MAMBAF405 target. Barometer is connected to standard I2C pads on the FC, is detected correctly and functions normally.

quetiano, 2020-04-07:
Yes , mines work with baro alone or compass alone, but never the both together on the same i2c pads unfortunately

MaxMihawk, 2020-04-07:
# INAV/MAMBAF405 2.4.0 Feb 11 2020 / 23:28:24 (xxx)

My dump.
[INAV_cli_Tyro129+_20200407_203129.txt](https://github.com/iNavFlight/inav/files/4446413/INAV_cli_Tyro129%2B_20200407_203129.txt)

> 
> 
> Make a photo of your wiring or mark it in the wiring sheme and post it here. Most of your questions are answered in the betaflight manual. Maybe you could read it... https://github.com/iNavFlight/inav/wiki/Navigation-modes

I'm starting to think you don't need any help. You only need a chat.

quetiano, 2020-04-07:
Hello MaxMihawk.
No way, I require all the help I can get.
I had help and advice from others as well and I also had my own problems today such as needed to resolder pins on the mamba because they got loose , I did not do a very good soldering pins, so did a proper job just now, and also performed my own tests, but none worked. Others that have also advised me was not good advice because I've already tried all what they have advised me to do. I am just trying something now before giving up, this is the final straw. I will get back to you once it works, the mamba is performing with bitch mode now, red lit led currently and i dont know why, once its sorted, I will try one more thing. 

--------------------------------------------------------------------------------------------------------
> # INAV/MAMBAF405 2.4.0 Feb 11 2020 / 23:28:24 (xxx)
>
> My dump.
> [INAV_cli_Tyro129+_20200407_203129.txt](https://github.com/iNavFlight/inav/files/4446413/INAV_cli_Tyro129%2B_20200407_203129.txt)
>
>
> I'm [...]

MaxMihawk, 2020-04-07:
> Thank you for your dump, it seems that the bmp 280 is better compatible with inav than my ms5611 baro.

https://github.com/iNavFlight/inav/wiki/Target-and-Sensor-support
I don´t think so.

quetiano, 2020-04-07:
Sure, I will do that tomorrow, the time is 22:38, going to get ready to go to bed. Thank you so much for your help, will reply tomorrow, after checking the mamba. Will check if any wires are touching as I did some soldering. Till tomorrow

quetiano, 2020-04-08:
Apologies everybody. Flight controller is gone faulty, did some quick soldering of the pins, then came back to flash a new firmware, the mamba red led is staying lit, no blinking, sometimes the pc picks it up and it flashes blue with the red steady led lot, but suddenly the FC gets disconnected. I tried a phone on the same cable and it worked fine, so it cant be the cable,has to be the FC. As of now, its just steady red led on. Maybe  I damaged something during soldering. Thats the end of me, I can only order another FC after the lockdown here which can be some time. Thanks to everybody for their help guys, much appreciated !!!

closed 2020-06-21 by stale[bot]