iNavFlight/inav #5496: Quad will flip over rather than compensate if one motor performs weakly

evranch, 2020-03-11:
## Current Behavior
The bearings are going bad on one motor on my 8" quad. At full throttle, the quad will rapidly roll towards that motor and can go into a tumble.

## Steps to Reproduce
1. Hover at 50% throttle
2. Fly around, no issues
3. Punch out at 100% throttle
4. Quad will roll towards weak motor
5. Recover from dangerous attitude

## Expected behavior
If one motor cannot keep up, power to the other motors should be reduced to keep the quad level.
Uncommanded attitude changes should not occur.

## Suggested solution(s)
If one motor is at 100% throttle but the quad is still tilting towards it, reduce overall throttle setpoint until there is enough headroom for a weak motor to level the quad. If this results in settling to the ground, this is safer than flipping over.

This issue does not occur when running Betaflight on the same quad with the same sick motor, it climbs slower but the attitude is not affected. I suspect Betaflight detects the roll and compensates by lowering thrust.

## Additional context

A typical event showing all motors at steady throttle despite 59deg/s of uncommanded roll
[image]

Please ignore the resonance in this log, I was pushing the PIDs to see what would happen. This issue occurs even when the quad is flying smoothly.
[blackbox_log_2020-03-10_191059_rollover.log](https://github.com/iNavFlight/inav/files/4316022/blackbox_log_2020-03-10_191059_rollover.log)

---

- FC Board name and vendor: HGLRC XJB F438-TX20 stack (Omnibus F4)
- INAV version string: INAV/OMNIBUSF4 2.4.0 Feb 11 2020 / 22:11:36 (bcd4caef9)

diff:

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gf

# feature
feature DYNAMIC_FILTERS
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 64 115200 115200 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 2100
aux 1 1 0 900 1350
aux 2 11 4 1300 2100
aux 3 17 6 1300 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 2 V
osd_layout 0 1 12 0 H
osd_layout 0 7 13 11 V
osd_layout 0 9 0 10 V
osd_layout 0 13 22 7 V
osd_layout 0 14 25 1 V
osd_layout 0 15 25 10 V
osd_layout 0 22 14 10 V
osd_layout 0 24 12 2 V
osd_layout 0 30 0 12 V
osd_layout 0 32 1 11 V
osd_layout 0 34 9 2 H

# master
set gyro_sync = OFF
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 70
set gyro_lpf_type = PT1
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = MPU6000
set acczero_x = -31
set acczero_y = 40
set acczero_z = 5
set accgain_x = 4095
set accgain_y = 4063
set accgain_z = 4100
set align_mag = CW0FLIP
set mag_hardware = QMC5883
set magzero_x = -36
set magzero_y = 78
set magzero_z = 141
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set throttle_idle =  9.000
set model_preview_type = 3
set applied_defaults = 2
set gps_sbas_mode = WAAS
set nav_extra_arming_safety = OFF
set osd_video_system = NTSC

# profile
profile 1

set mc_p_pitch = 80
set mc_i_pitch = 100
set mc_d_pitch = 18
set mc_p_roll = 80
set mc_i_roll = 100
set mc_d_roll = 18
set mc_p_yaw = 95
set mc_p_level = 8
set mc_i_level = 10
set use_dterm_fir_filter = OFF
set yaw_lpf_hz = 30
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set rate_accel_limit_roll_pitch = 3600
set rate_accel_limit_yaw = 720
set heading_hold_rate_limit = 30
set d_boost_factor =  1.500
set d_boost_gyro_delta_lpf_hz = 40
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set thr_expo = 70
set roll_rate = 40
set pitch_rate = 40

# battery_profile
battery_profile 1

Quick-Flash, 2020-03-12:
This should be more of a pid tuning issue. When one motor output maxes out (aka one motor is weaker than the rest) the mixer will limit the output of the other motors. If properly tuned it should act as you said, which is probably why it is fine when using betaflight

evranch, 2020-03-12:
OK, so this is already the expected behaviour of the mixer. Which of the PID terms should be adjusted to allow it to work correctly? As seen in the image, the other motors initially drop in output slightly but the roll continues.

I can keep the weak motor on and do some testing to see if this can be tuned out.
I have already varied P and I from 30 to 100 with no difference in the rolling behaviour.

Note that I recently upgraded iNav and that is when this issue started on the first flight. Previous version was 2.1.0, and it was flying great with the default P and D for 7" endurance and 35 I.

DzikuVx, 2020-03-21:
This is not how PID controller and mixer work. Flight controller is not able to determine if a disturbance (tumble to the side) is caused by a external factor (wind gust) or lower power output on one of the motors. 

Also, there is no direct feedback loop between FC, ESC and motor. FC asks: `set power to 70%` but has no way to know is ESC put that power level or motor was able to convert it into thrust. There is only indirect information from gyro that says if stability is in place or not. 

On top of that, there is a conflict between user demanding throttle and stability. System will be stable as long as all the actuators (motors and props) have similar output level. If one of the actuators max output is smaller than the other, system will be stable only as long as output demand will be below this level.

With all of that in place: flight controller is not aware of all the data required to lower throttle when one of the actuators saturates. The fact that one of the outputs is saturated does not mean that actuator is saturated. In some cases those are the same events, in other are different.

This is a very much an edge case that can not be solved cheap, fast and in an reliable way.

closed 2020-03-21 by DzikuVx