iNavFlight/inav #5506: Hi guys, I have a question about RSSI. I am using fport with 405wing and 765wing FC. The RSSI show around 15% different from OSD and lua screen on the radio. what is the reason to cause it and how to calibrate it?

xixijoe, 2020-03-14:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)
* [How to setup INAV on a flying wing](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLn_lCX7UB2OPLzR4w5G4el3)
* [How to setup INAV on a 5" quad](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLl9M6Vm-ZNLMCr6_9yNVHQG)

ShikOfTheRa, 2020-03-17:
For Fport, normally have to calibrate. Teh follwoing is documented in many places. This is what I did too.

set rssi_scale = 200
save

xixijoe, 2020-03-17:
> For Fport, normally have to calibrate. Teh follwoing is documented in many places. This is what I did too.
> 
> set rssi_scale = 200
> save

this command is not long available in the CLI. can you check it? I guess it is for the old version of Inav

thesync, 2020-03-23:
For F.Port you should use the following commands.
set rssi_source = PROTOCOL
set rssi_max = 50

F.Port RSSI max value is 50. Using these commands you tell the FC that 50 is the maxo so it will scale 0-50 to a value of 0-100

xixijoe, 2020-03-27:
> For F.Port you should use the following commands.
> set rssi_source = PROTOCOL
> set rssi_max = 50
> 
> F.Port RSSI max value is 50. Using these commands you tell the FC that 50 is the maxo so it will scale 0-50 to a value of 0-100

it works. great! thank you

closed 2020-03-27 by xixijoe