iNavFlight/inav #5507: Mamba F405 servo output remap

purpletreee, 2020-03-14:
Hello guys, 
I have a mamba f405 (https://www.diatone.us/products/mamba-f405-flight-controller-mk%E2%85%B1) and I need to have a servo output. But there isnt a special pin for servo output as you can see and the 4 motor out pins are all used because it is in a drone. Nevertheless in the inav configurator there are a lot resources free 
`# resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: MOTOR4 OUT
A03: MOTOR1 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: BEEPER OUT
A09: FREE 
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FREE 
B00: MOTOR2 OUT
B01: MOTOR3 OUT
B02: FREE 
B03: FLASH CS
B04: LED2 OUT
B05: LED1 OUT
B06: I2C1 SCL
B07: I2C1 SDA
B08: FREE 
B09: FREE 
B10: FREE 
B11: FREE 
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: INVERTER OUT
C01: ADC CH1
C02: FREE 
C03: ADC CH2
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D02: FREE 
`. So I am asking if anybody could make a new target to use any of those spare pins as a servo/pwm output.

Thank you in advance :)

purpletreee, 2020-03-14:
or any of  those beeper or led pins can be replaced aswell.

purpletreee, 2020-03-14:
If someone explaines me how to do it, I can do it myself aswell

Saschko2000, 2020-05-03:
I have the same problem. Can nobody help?

DzikuVx, 2020-05-27:
There are no other broken out pins that allows to be used as a servo. This is a quadcopter in mind designed flight controller

closed 2020-05-27 by DzikuVx

fremebondo, 2020-05-30:
And what if I want to use LED pin to drive a servo?

njyouzhiyong, 2023-01-11:
you can use ppm/C09, like "resource SERVO 1 C09"

streem-ua, 2023-03-23:
Hello! @purpletreee Have you figured out how to connect a servo? I have mamba f722 mk2 and in same situation

MagistrK, 2024-06-12:
> ви можете використовувати ppm/C09, наприклад "ресурс SERVO 1 C09"

Чи є ще варіанти окрім цього ?