iNavFlight/inav #5510: Autolaunch throttle voltage based settings

brassman84, 2020-03-17:
## Current Behavior
The autolaunch throttle setting is universal and has to be changed depending on what cell count you are using. A launch throttle calibrated for 6s will crash the plane if you use a 4s.
I like to fly fast with a small 6s pack but sometimes I want to have longer flight times and go with a higher mah 4s pack.

## Desired Behavior
The possibility to set different launch throttle values for 3s,4s,5s and 6s. Not automatic values scaled by voltage but  four actual launch throttle settings manually calibrated for each cell count and used depending on the readings from the automatic cell detection feature already in place

## Suggested Solution
Always active, default value should be the same for all cell counts to not cause any problems that have issues with the automatic cell count detection

## Who does this impact? Who is this for?
Airplanes only, anyone who uses autolaunch and want to run different cell counts in one plane.

peteoz, 2020-03-17:
if this is considered then it would also make sense to allow variable cruise throttle for navigation modes based on number of cells we are running.

for example on my Nano Goblin I run 2S, 3S, 4S & 5S batteries, every time I change between different battery type I need to go to iNav settings and modify  "nav_fw_launch_thr" as well as "nav_fw_cruise_thr" values.

that means dragging along laptop every time I go fly.

brassman84, 2020-03-17:
Thats a great idea Peteoz👍I too have the same issue with using cruise mode

brassman84, 2020-03-17:
Also just to clearify the reason why you dont want a automatic scaling for this stuff, when switching from 4s to 6s you most likely have to switch to a different propeller too

swissfreek, 2020-03-21:
Not necessarily. Would be nice to have an automatic switcher like Betaflight does so you can fly with either pack and not have to change a bunch of settings.

peteoz, 2020-03-21:
from 2S to 5S I don't change propeller

closed 2020-06-03 by stale[bot]